OctoPrint/OctoPrint #3292: Action Command Prompt: Rendering error with long texts

leptun, 2019-09-29:
#### What were you doing?

I was implementing a "Print from SD" with button with full filename structure into Prusa-Firmware that used Action Command Prompt as the plugin to display the files, but ran into rendering issues with it when the text in the button became too long. You can see it in the attached photo.

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

This is using one of the built-in plugins and requires it to be enabled.
#### Version of OctoPrint

OctoPrint 1.3.11
#### Operating System running OctoPrint

Windows 10
#### Printer model & used firmware incl. version

Not applicable, but it's a Prusa Mk3S
#### Browser and version of browser, operating system running browser

N/A. Happens in all browsers. The browser is where the host is located.
#### Link to octoprint.log
https://gist.github.com/leptun/ff3f1cfda1fe534b081f4ebf49e035db

#### Link to contents of terminal tab or serial.log
https://gist.github.com/leptun/234d887458210058d3cfc1c52d3d7b2a

#### Link to contents of Javascript console in the browser
https://gist.github.com/leptun/ae5ef70c34e1265d8915df5ac2a6633d

#### Screenshot(s)/video(s) showing the problem:

https://photos.app.goo.gl/xVqXAvABzW23c3Q26
[image]

I have read the FAQ.

foosel, 2019-10-08:
Admittedly I never anticipated text as long as that and hence used a simple button here which doesn't wrap text on its own. Guess that needs to be extended slightly to accommodate use cases like this one.

leptun, 2019-10-08:
Thank you for looking into this. :)

foosel, 2019-10-25:
[image]

[image]

Implemented by the above commit, will be in the next release.

leptun, 2019-10-25:
Thank you! Closing issue.

closed 2019-10-25 by leptun