OctoPrint/OctoPrint #3293: [Request] Add RPi temperatures to the graph in the temperatures page

LinoBarreca, 2019-10-01:
When Octoprint is running on Raspberry PI it would be nice to have in the "temperatures' graph also the temperature of the CPU and the GPU of the RPi itself.
It can get pretty hot when printing or encoding webcam.

Actually I have to monitor it through the Dashboard plugin (but there's no graph)

louispires, 2019-10-15:
This would be awesome - Another great addition to the Temperature graph would be to scroll horizontally back in time.

LinoBarreca, 2019-10-15:
> 
> 
> This would be awesome - Another great addition to the Temperature graph would be to scroll horizontally back in time.

you can actually configure the horizontal resolution of the graph in the settings.

closed 2019-10-30 by foosel