OctoPrint/OctoPrint #3294: Bug: Octoprint 1.3.12rc3 doesn't start on Buster

LinoBarreca, 2019-10-02:
#### What were you doing?
SSH to OctoPI

#### What did you expect to happen?
octoprint running.

#### What happened instead?

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/__init__.py"                                                                                                   , line 6, in <module>
    import logging as log
  File "/usr/lib/python2.7/logging/__init__.py", line 26, in <module>
    import sys, os, time, cStringIO, traceback, warnings, weakref, collections
  File "/usr/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

#### Did the same happen when running OctoPrint in safe mode?
it doesn't start at all.

#### Version of OctoPrint
OctoPrint 1.3.12rc3 running on OctoPi 0.16.0

#### Operating System running OctoPrint
Linux octopi 4.19.75-v7l+ #1271 SMP Mon Sep 30 13:51:15 BST 2019 armv7l GNU/Linux
(buster)

#### Printer model & used firmware incl. version
not applicable

#### Browser and version of browser, operating system running browser
not applicable

#### Link to octoprint.log
nothing on log. octoprint doesn't start

#### Link to contents of terminal tab or serial.log
message is above

#### Link to contents of Javascript console in the browser
error not on browser but in the terminal running octoprint

#### Screenshot(s)/video(s) showing the problem:
none. message is above.

I have read the FAQ.

foosel, 2019-10-03:
Since OctoPi 0.16 doesn't ship with Buster I take it you updated this image. How, what exact steps? Do you get the same issue with a 0.17 build (e.g. [this one](https://community.octoprint.org/t/raspberry-pi-4-is-out/10332/8?u=foosel))?

LinoBarreca, 2019-10-03:
Okay you need the whole history to understand what happened:
pi3b+ with octopi 0.16 standard
Had to upgrade to pi4 and, for that I needed the new kernels.
I resized the boot partition on the sd.
On the pi3 I did a sudo rpi-update. This updated the kernel on the boot partition.
Put the card on the pi4. booted fine and everything was working.
changed octoprint's update channel, it offered me the rc3. I updated, rebooted and verified everything was working.

Then I thought "let's see if the 1.3.12 works on buster" so 

grep -rl stretch /etc/apt/ | sudo xargs sed -i 's/stretch/buster/g'
to change the distro and
sudo apt update && sudo apt dist-upgrade -yy && sudo apt full-upgrade

to upgrade it.
rebooted and the pi (along with its program) worked fine....octoprint doesn't start instead. so i did the SSH and at the ssh login, instead of giving me the version installed, as it usually does, it gives me the error above.
Same if I start it through command line.

foosel, 2019-10-07:
Hm... that doesn't look like a general problem with Buster, I'm sorry to say. 

Looking at the data in the anonymous usage tracking I can see 48 instances that have been seen over the past 30 days, running any of the three 1.3.12 RCs under OctoPi 0.17.0 nightly, which is a Buster build. Also with a healthy mix across various Raspberry Pi versions.

My money would rather be on something in the dist upgrade breaking something crucial - frankly, I've never trusted this on the Raspbian builds myself, it has been hit and miss in my experience.

The stack trace also looks like something about the Python environment is fundamentally broken, outside of anything that OctoPrint has any kind of control over. Note how it crashes on import of the logging library, which is a core Python library.

I tried to reproduce this error by doing the steps which made it break for you, and it got it to break but with a different error:

```
(oprint) pi@octopi2:~ $ octoprint serve --debug
Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 74, in <module>
    __import__('pkg_resources.extern.packaging.requirements')
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pkg_resources/extern/__init__.py", line 61, in load_module
    "distribution.".format(**locals())
ImportError: The 'packaging.requirements' package is required; normally this is bundled with this package so if you get this warning, consult the packager of your distribution.
```

My guess is that the virtualenv somehow got nuked by the dist upgrade. And lo and behold, recreating the dist upgrade and installing 1.3.12rc3 into it makes everything work again:

```
mv ~/oprint ~/oprint.old
virtualenv ~/oprint
source ~/oprint/bin/activate
pip install https://github.com/foosel/OctoPrint/archive/1.3.12rc3.zip
```

So... yeah. It seems that dist-upgrade can kill existing virtual environments. That's completely outside of the control of OctoPrint however. If at all it would be an OctoPi topic, but to be frank I rather think it's a Raspbian and/or virtualenv topic.

closed 2019-10-07 by foosel

lostapathy, 2019-10-23:
FYI @foosel this must not be totally isolated to OP's pi - the same upgrade just did the same thing to me.  Your prior comment did allow me to get it going again, thanks for posting that.

foosel, 2019-10-31:
For the record, I didn't say it's a problem isolated to the OP's Pi, I said it's isolated to the manual upgrade path from Jessie to Buster (and not a general issue with Buster) and apparently caused by something in that upgrade path breaking existing virtual environments.

thhart, 2019-11-01:
Hi Gina, you are right, a dist upgrade is out of scope of any venv, but what strikes me was the matter of fact there was no error log anywhere. I only could see the error above when I started octoprint server from the console. The service simply terminated, no message, no log anywhere. Maybe this is something to improve...

foosel, 2019-11-01:
And how am I supposed to log anything from a piece of Python code if the Python environment it is supposed to run in is completely borked and refuses to run anything? ;)

thhart, 2019-11-01:
Sorry - I am not in python and see it from an alien point of view. But somehow the systemd service runner should be able to catch even from python failing this early and be it the correct exit code. I wish I would have time to dive into the service file, but I can not do now. It was just an idea, drop it and keep rocking with the rest. ;-)

tablatronix, 2020-08-29:
reinstalling from same issue, error I saw was

```
octoprint ImportError: cannot import name _remove_dead_weakref
```

fieldOfView, 2020-08-29:
Googling your error message, I came across this:
https://askubuntu.com/questions/981663/python2-7-broken-by-weakref-import-error-please-help
You may have to recreate your virtualenv

FranPalmer, 2021-02-25:
OK, need some external verification, but I just seemed to fix this.

I performed the raspberian upgrade to prepare for moving this octopi install to a pi4 from a pi3 in the near future, and ran into this same error.

I went into raspi-config, and I generated the localization files for both us-EN.UTF-8 and us-GB.UTF-8.   I also updated the time zone.  The pi rebooted when I exited raspi-config and octoprint started up normally.

I will be doing this upgrade again shortly (waiting on a print to finish) and will attempt to verify this myself.