OctoPrint/OctoPrint #3295: Updater confused by running development version

fake-name, 2019-10-04:
1. `virtualenv venv`
 2. `source venv/bin/activate` 
 3. `pip install git+https://github.com/foosel/OctoPrint.git@devel`
 4. Run octoprint
 5. Go to webpage

-> 

```
Updating, please wait.
+++++++++++++++++++++++++++
Now updating None to 1.3.11
+++++++++++++++++++++++++++
/usr/local/octoprint/venv/bin/python3 -m pip --disable-pip-version-check install https://github.com/foosel/OctoPrint/archive/1.3.11.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/foosel/OctoPrint/archive/1.3.11.zip
Downloading https://github.com/foosel/OctoPrint/archive/1.3.11.zip
ERROR: Package 'OctoPrint' requires a different Python: 3.7.3 not in '>=2.7.9,<3'
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
```

Environment

```
130 pi@raspberrypi:/usr/local/octoprint$ source venv/bin/activate
(venv) pi@raspberrypi:/usr/local/octoprint$ python --version
Python 3.7.3
(venv) 130 pi@raspberrypi:/usr/local/octoprint$ uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
(venv) pi@raspberrypi:/usr/local/octoprint$ cat /etc/issue
Raspbian GNU/Linux 10 \n \l

```

It seems like the updater should be disabled in the dev version, since it apparently doesn't produce sane results.

fake-name, 2019-10-04:
Wow, that bot is annoying.

---

Apparently there is *kind of* a mechanism for handling dev installs. 

[image]

However, I installed via `pip git+https`, which means there is no local checkout, and *you can't turn off updates either*.  Putting a bad path in as the `OctoPrint checkout folder` shuts the updater up, but I don't know if it'll break anything else.

FWIW, a python 3 release would basically solve this. I started dealing with this because multiple plugins in the normal plugin repository apparently depend on updated numpy, which doesn't work anywhere other then python 3.5 or later.

fake-name, 2019-10-04:
So, if a plugin is incompatible, that fact is not reported to the UI. Like, **at all.**

They look like they're installed correctly, but just don't do anything.  That's super confusing.

jneilliii, 2019-10-05:
It writes output to octoprint.log on incompatibility of plugins.  When you first load OctoPrint after install just click ignore on the update prompt.  I actually have octoprint checked out in order to be able to track commit updates and automatically update the checkout when changes are incorporated upstream on the devel branch.

fake-name, 2019-10-05:
> It writes output to octoprint.log on incompatibility of plugins. 

Yeah, I found that after being confused that plugins that showed up in the web UI were mysteriously not working.

I wound up patching [`pythoncompat`](https://github.com/foosel/OctoPrint/blob/devel/src/octoprint/plugin/core.py#L435-L445) to assume compatibility, and everything started working (well, except that my printer is stupid and I hit https://github.com/foosel/OctoPrint/issues/2309 because it has a ChiTu controller).

jneilliii, 2019-10-05:
You can actually just add in the following in your config.yaml and allow you to force the plugins to try and load. If they aren't compatible they could still error and not load.
```
plugins:
  _forcedCompatible:    
  - bedlevelvisualizer
```

fake-name, 2019-10-05:
Well, it's a moot point now.

foosel, 2019-10-05:
> It seems like the updater should be disabled in the dev version, since it apparently doesn't produce sane results.

The real issue here is that something with the version number detection is broken for some reason, it should say 1.4.0.dev... there, not 0. I also have a local reproduction of that against a git checkout. So that will be looked into.

> FWIW, a python 3 release would basically solve this. I started dealing with this because multiple plugins in the normal plugin repository apparently depend on updated numpy, which doesn't work anywhere other then python 3.5 or later.

Yeah, and I'm working on that.

In the meantime it would be nice to open reports in affected plugin repositories because apparently they should have pinned a version there in their dependencies and didn't.

> Wow, that bot is annoying.

So is having to constantly run after logs, reproduction steps and so on - for almost seven years now 🤷‍♀️ If you have a better idea to solve that and still be able to actually work on code (e.g. to push out releases with compatibility fixes) and not just triage reports more or less full time, I'm open for suggestions. As things are, this is a good approach for things to stay manageable.

foosel, 2019-10-05:
Also as a general heads-up: a bit less "urgh this" and "urgh that" would go a long way regarding the tone of interaction and attitude on display here so far by you @fake-name. Thank you.

fake-name, 2019-10-05:
> Also as a general heads-up: a bit less "urgh this" and "urgh that" would go a long way regarding the tone of interaction and attitude on display here so far by you @fake-name. Thank you.

Sorry. I need to stop bug reporting when angry, (though I thought that was only about discourse and that's because it breaks ctrl + f and).

> I'm open for suggestions. As things are, this is a good approach for things to stay manageable.

Some mechanism for saying "This issue doesn't fit the template" should be more then adequate, I think?

------

The root of this is that multiple plugins depend on `numpy`, and there's something odd going on about how the plugin dependency processing works, so that they don't install the last Py2K version. It instead fails to install altogether.

jneilliii, 2019-10-06:
One of my plugins (bedlevelvisualizer) uses numpy and installs on my windows dev box without errors in python 3 environment as well. That was one of my hopes of the python 3 compatibility was because I thought numpy was delivered compiled versus it having to compile during install, which has caused many headaches for users in the past.

fake-name, 2019-10-06:
To be fair, once I fiddled with the compatibility flags to convince it to load, it seems to be fine (well, except my printer's bed leveling doesn't seem to work properly).

I *think* raspbian may not be shipping py2 numpy for ARM at this point, but I have no idea what was going on.

foosel, 2019-10-06:
That should be fixed now by the above commits. I also added some more configuration options to the Software Update Plugin so you can now select github_commit tracking right from the UI and configure the pip target URL to use, so you *should* be able to make your setup work by selecting Github Commit tracking, configure the devel branch and then change the pip target to your install URL. At least I hope so, I haven't yet gotten around to testing that particular setup (and frankly it's a bit unusual even though valid - usually you either have Github releases or a full checkout).

closed 2019-10-06 by foosel

fake-name, 2019-10-06:
Fancy!

> and frankly it's a bit unusual even though valid - usually you either have Github releases or a full checkout

I dunno about octoprint users specifically, but I use git+https installs extensively when doing python stuff, mostly because it's super easy. You can even put  them in a `requirements.txt` file. If you depend on packages that don't have up-to-date PyPi releases, it can totally save your bacon.

foosel, 2019-10-07:
I was talking specifically about the OctoPrint community ;)

commit cce7389a (2019-10-06): ✨ SWU: More configuration options

Allow to select github_commit tracking, branch and pip target via UI.

Closes #3295