OctoPrint/OctoPrint #3298: [Minor] Loading animation hides the login screen forever when the server errors

fake-name, 2019-10-08:
Running the devel version of octoprint from master. 

The loading animation never goes away if the `http://<printer>/api/login` page returns a 500 error.

I am aware I have other issues (the server returning a 500), but it's super confusing. 

First, loading animations on *web pages* are completely absolutely utterly terrible and shouldn't be a thing. If you really, really have some reason for it, please let me turn it off permanently somehow (in a config?).

It also makes it very hard to determine that something *isn't* working. Instead of an error, you just get the animation spinning forever. If you have to have a loading animation, there's a common standard of having the wrapper that hides the page go away after 8-10 seconds in all cases, so situation like this (where things go wrong) end with *something* happening.

This is a pretty minor thing, and it is on the development branch.

foosel, 2019-10-24:
Implemented by the above commit.

closed 2019-10-24 by foosel

fake-name, 2019-10-25:
Woot!