OctoPrint/OctoPrint #3299: config.yaml - Garbage Collection

LMS0815, 2019-10-08:
Any installed plugin leaves its footprint in _config.yaml_ although it is/was disabled and uninstalled.

Please provide an option to clean _config.yaml_ from all unused plugin references (_disabled flag and configuration lines).

Please consider providing an option to clean associated references during uninstall of  a plugin.

foosel, 2019-11-01:
Implemented by the above commit.

Can be triggered on uninstall:
[image]

Or manually:
[image]

Will be part of the next release.

LMS0815, 2019-11-07:
Thank you. 
If possible add an "after the fact" / global cleanup button to the plugin manager too.
Would be handy, to ensure, that there are no leftovers form old installations and or you missed to clear during uninstall.

foosel, 2019-11-15:
[image]

Full cleanup feature is now there as well.

LMS0815, 2019-11-18:
THANK YOU SO MUCH!!! - Awaiting the new release.

foosel, 2020-03-04:
1.4.0 has been released.

closed 2020-03-04 by foosel

LMS0815, 2020-03-06:
Thank you. Just a small amendment. Would be nice to have an option to clear configuration of installed plugins to revert to defaults...