OctoPrint/OctoPrint #3300: [DEV] {Python3] Issue with __plugin_settings_overlay__

jneilliii, 2019-10-08:
## What were you doing?

Verifying compatibility with Python 3 and OctoPrint 1.40 devel branch with a plugin that has `__plugin_settings_overlay__` for ordering tabs causes a hard error to core OctoPrint and is unable to be loaded.

#### What did you expect to happen?

Plugin to work the same as it does in Python 2 environment.

#### What happened instead?

Server fails to load completely.

#### Did the same happen when running OctoPrint in safe mode?

N/A since it's specific to a plugin feature.

#### Version of OctoPrint

OctoPrint 1.4.0.dev2375+g09a4fb25

#### Operating System running OctoPrint

Windows and Python 3.7.4

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Google Chrome Version 77.0.3865.90 (Official Build) (64-bit)

#### Link to octoprint.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/3703965/octoprint.log)

#### Link to contents of terminal tab or serial.log

Error is not captured in octoprint.log, but this is the message displayed in the command window when running `octoprint serve`.

```
2019-10-08 14:14:51,399 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://192.168.0.113:5000/', 'path': '/', 'query_string': 'l10n=en'}
[2019-10-08 14:14:51,556] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "c:\octoprint\venv\lib\site-packages\flask\_compat.py", line 33, in reraise
    raise value
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "c:\octoprint\venv\lib\site-packages\octoprint\server\views.py", line 181, in index
    fetch_template_data(refresh=force_refresh)
  File "c:\octoprint\venv\lib\site-packages\octoprint\server\views.py", line 755, in fetch_template_data
    sorted_missing = sorted(missing_in_order, key=key_func)
TypeError: '<' not supported between instances of 'str' and 'NoneType'
[2019-10-08 14:14:51,743] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "c:\octoprint\venv\lib\site-packages\flask\_compat.py", line 33, in reraise
    raise value
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "c:\octoprint\venv\lib\site-packages\octoprint\server\views.py", line 181, in index
    fetch_template_data(refresh=force_refresh)
  File "c:\octoprint\venv\lib\site-packages\octoprint\server\views.py", line 755, in fetch_template_data
    sorted_missing = sorted(missing_in_order, key=key_func)
TypeError: '<' not supported between instances of 'str' and 'NoneType'
[2019-10-08 14:14:51,884] ERROR in app: Exception on / [GET]
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "c:\octoprint\venv\lib\site-packages\flask\_compat.py", line 33, in reraise
    raise value
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "c:\octoprint\venv\lib\site-packages\flask\app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "c:\octoprint\venv\lib\site-packages\octoprint\server\views.py", line 181, in index
    fetch_template_data(refresh=force_refresh)
  File "c:\octoprint\venv\lib\site-packages\octoprint\server\views.py", line 755, in fetch_template_data
    sorted_missing = sorted(missing_in_order, key=key_func)
TypeError: '<' not supported between instances of 'str' and 'NoneType'
```

#### Link to contents of Javascript console in the browser

N/A

I have read the FAQ.

foosel, 2019-10-10:
This should be fixed by the above commit (`sorted` or rather its custom key functions changed behaviour in 3.7, are no longer `None` tolerant and that was overlooked during initial refactoring). Could you give it a test?

foosel, 2019-10-10:
And for some reason that has broken the 2.7 unit tests. Cannot reproduce that locally at the moment, will possibly have to wait until next week. Seems to be more an issue with the unit test however than the code.

jneilliii, 2019-10-10:
I can confirm this is resolved in Python 3 environment.

foosel, 2019-10-11:
Perfect, thank you! Will close once I've fixed the tests. Which get weirder and weirder, yesterday only Py2.7 was failing, now it's also some of the Py3 versions, locally all is fine... It's really weird. I have a 4h train ride later today though, so maybe I'll be able to figure something out then.

foosel, 2019-10-14:
Well, that should also be solved now.

commit a889bec4 (2019-10-10): 🐛 Solve sorted issues under Python 3.x

sorted does no longer support sorting None values under Python 3. This
patch introduces a utility function that ensures that None values are
replaced by a configurable default value ("" by default) and utilizes
it in places where sorted gets called with a custom sorting key that
can potentially consist of None values.

Closes #3300
closed 2019-10-14 by foosel