OctoPrint/OctoPrint #3301: [Bug] Race condition in SD print detection can cause just cancelled print to "start" again

foosel, 2019-10-09:
### Problem

See https://github.com/foosel/OctoPrint/issues/3125#issuecomment-539587212 and https://github.com/foosel/OctoPrint/issues/3125#issuecomment-539879264 as well as

```
2019-10-08 18:03:38,625 - Changing monitoring state from "Printing from SD" to "Cancelling"
2019-10-08 18:03:38,684 - Send: M25
2019-10-08 18:03:39,043 - Recv: SD printing byte 289651/383498
2019-10-08 18:03:39,044 - Recv: ok
2019-10-08 18:03:39,047 - Recv: ok
2019-10-08 18:03:39,048 - Send: M27
2019-10-08 18:03:39,052 - Send: M26 S0
2019-10-08 18:03:39,052 - Recv: SD printing byte 289651/383498
2019-10-08 18:03:39,054 - Recv: ok
2019-10-08 18:03:39,057 - Send: M84
2019-10-08 18:03:39,057 - Recv: !! SD busy
2019-10-08 18:03:39,093 - Recv: ok
2019-10-08 18:03:39,096 - Send: M104 T0 S0
2019-10-08 18:03:39,107 - Send: M140 S0
2019-10-08 18:03:39,379 - Send: M27
2019-10-08 18:03:40,382 - Send: M27
2019-10-08 18:03:40,828 - Recv: ok
2019-10-08 18:03:40,833 - Send: M106 S0
2019-10-08 18:03:40,835 - Recv: ok
2019-10-08 18:03:40,840 - Recv: ok
2019-10-08 18:03:40,843 - Changing monitoring state from "Cancelling" to "Operational"
2019-10-08 18:03:40,844 - Recv: SD printing byte 289651/383498
2019-10-08 18:03:40,848 - Changing monitoring state from "Operational" to "Starting print from SD"
2019-10-08 18:03:40,882 - Send: M25
2019-10-08 18:03:40,884 - Send: M27
2019-10-08 18:03:40,897 - Recv: ok
2019-10-08 18:03:40,908 - Recv: SD printing byte 289651/383498
2019-10-08 18:03:40,918 - Recv: ok
2019-10-08 18:03:40,923 - Send: M26 S0
2019-10-08 18:03:40,939 - Recv: ok
2019-10-08 18:03:40,943 - Send: M105
2019-10-08 18:03:40,952 - Send: M84
2019-10-08 18:03:40,960 - Recv: ok
2019-10-08 18:03:40,973 - Recv: SD printing byte 289651/383498
2019-10-08 18:03:40,976 - Send: M104 T0 S0
2019-10-08 18:03:40,991 - Recv: ok
2019-10-08 18:03:40,995 - Recv: ok
2019-10-08 18:03:41,000 - Send: M140 S0
2019-10-08 18:03:41,008 - Recv: ok B:60.0 /0.0 T0:209.9 /0.0
2019-10-08 18:03:41,031 - Send: M106 S0
2019-10-08 18:03:41,036 - Changing monitoring state from "Starting print from SD" to "Operational"
2019-10-08 18:03:41,038 - Recv: ok
2019-10-08 18:03:41,048 - Changing monitoring state from "Operational" to "Printing from SD"
2019-10-08 18:03:41,050 - Recv: ok
2019-10-08 18:03:41,063 - Recv: ok
2019-10-08 18:03:41,065 - Recv: ok
2019-10-08 18:03:41,383 - Send: M27
2019-10-08 18:03:41,388 - Recv: SD printing byte 0/383498
2019-10-08 18:03:41,390 - Recv: ok
2019-10-08 18:03:42,387 - Send: M27
2019-10-08 18:03:42,399 - Recv: SD printing byte 0/383498
2019-10-08 18:03:42,403 - Recv: ok
[...]
2019-10-08 18:03:58,289 - Changing monitoring state from "Printing from SD" to "Cancelling"
2019-10-08 18:03:58,358 - Send: M25
2019-10-08 18:03:58,378 - Recv: ok
2019-10-08 18:03:58,382 - Send: M27
2019-10-08 18:03:58,386 - Recv: SD printing byte 0/383498
2019-10-08 18:03:58,388 - Recv: ok
2019-10-08 18:03:58,391 - Send: M26 S0
2019-10-08 18:03:58,392 - Recv: ok
2019-10-08 18:03:58,395 - Send: M84
2019-10-08 18:03:58,400 - Recv: ok
2019-10-08 18:03:58,402 - Send: M104 T0 S0
2019-10-08 18:03:58,409 - Recv: ok
2019-10-08 18:03:58,413 - Send: M140 S0
2019-10-08 18:03:58,421 - Recv: ok
2019-10-08 18:03:58,424 - Send: M106 S0
2019-10-08 18:03:58,426 - Send: M27
2019-10-08 18:03:58,428 - Recv: ok
2019-10-08 18:03:58,431 - Changing monitoring state from "Cancelling" to "Operational"
2019-10-08 18:03:58,431 - Recv: SD printing byte 0/383498
2019-10-08 18:03:58,442 - Recv: ok
2019-10-08 18:03:59,427 - Send: M27
2019-10-08 18:03:59,439 - Recv: SD printing byte 0/383498
2019-10-08 18:03:59,445 - Recv: ok
```

### Solution

Find a way to detect if we just came out of Cancelling into Operational state when we see a SD printing message from an earlier M27 with a non zero current file position.

### Further information

First report in #3125 but actually a separate issue.

foosel, 2019-10-30:
@KevinOConnor pinging you here for a quick question since this bug seems to only be possible to reliably trigger on Klipper. To quote the above fixing commit:

> The problem seems to have been made worse by `M25` not being blocking
> on Klipper and thus the SD printing not being guaranteed to have stopped
> yet after its return, meaning the following `M26 S0` will be denied,
> causing the file position to not be properly reset and hence the issue
> to arise in the first place.

I took a look into the Klipper sources and noticed that `M25` will only set a flag to tell the virtual sd print runner to stop, it will not wait for it to actually stop. That's a bit of a behaviourly difference from other firmwares (or I've missed something completely the past years), because it means on Klipper I cannot be sure that the SD print has actually stopped after `M25` has been acknowledge by it. Is there some way to ensure that the print has stopped before proceeding, some parameter, some additional command? 

As things are right now, that puts me in a big of a tricky situation because I cannot depend on the state change to actually have gone through, meaning the states between OctoPrint and Klipper might not match any more and all kind of assumptions to not hold true (e.g. that I can reset the file position to 0 in case of a print cancel, which will fail and in the bug potentially trigger the issue this ticket is about).

KevinOConnor, 2019-10-30:
> I took a look into the Klipper sources and noticed that M25 will only set a flag to tell the virtual sd print runner to stop, it will not wait for it to actually stop. That's a bit of a behaviourly [...]

Okay, thanks.  That sounds like something that should be fixed in Klipper.

-Kevin

KevinOConnor, 2019-10-30:
This was a simple change to Klipper, but I'm having difficulty testing it because I seem to be hitting separate issues.  Perhaps you could give some pointers to assist.

Currently, Klipper reports the file position on an M27 command even after an M25.  So, for example, if the M25 came when printing at file offset 43 then M27 would report something like "SD printing byte 43/1989".  It's unclear if OctoPrint would prefer receiving "Not SD printing." instead.  I can make that change if it is preferable.

The second (and much larger issue) is that OctoPrint is getting out of sync with Klipper acknowledgements.  So, although the M25 now waits for the command to complete, I fear OctoPrint may still have an issue because it may interpret a previous M27 command as having arrived after the M25 due to the lost synchronization.

Here's the logs:
[octoprint.log](https://github.com/foosel/OctoPrint/files/3791310/octoprint.log)
[serial.log](https://github.com/foosel/OctoPrint/files/3791311/serial.log)

At first glance, it seems that OctoPrint is sending multiple "N0 M110 N0*125" commands at the start of the connection, but it is not expecting multiple "ok" responses?  Klipper follows a simple rule for acknowledgements - for every newline character (ascii 0x10) read it will respond with exactly one "ok" response.

The third issue is that after cancelling or pausing an sdcard print, OctoPrint continues to send an endless stream of M27 commands.  (It does this even if Klipper responds with "Not SD printing.".)  The log above shows this.  I'm not sure if that is expected behaviour or not.

-Kevin

foosel, 2019-10-31:
> Currently, Klipper reports the file position on an M27 command even after an M25. So, for example, if the M25 came when printing at file offset 43 then M27 would report something like "SD printing [...]

As far as I understand it from other firmwares (and that's the only thing I can go on here because as you know we don't have a proper well defined spec for this stuff 😭), `Not SD printing`  will only be returned if no SD file is selected (though not necessarily, sometimes it's also `SD printing 0/0`) and maybe if the SD print got cancelled through the printer's UI but the file is still selected. So OctoPrint reacts to that message with a state synchronization to "not printing" as well. If you start returning that when having been *paused*  by `M25` (we don't have a cancel command, so `M25` is used for both pausing and cancelling a print by OctoPrint, in the latter case also resetting the position to 0 via a follow-up `M26 S0`) it will cause issues, so simply continuing to respond with `SD printing byte <current>/<total>` should be totally fine here.

> The second (and much larger issue) is that OctoPrint is getting out of sync with Klipper acknowledgements.
> [...]
> At first glance, it seems that OctoPrint is sending multiple "N0 M110 N0*125" commands at the start of the connection, but it is not expecting multiple "ok" responses?

Yeah. The problem here is that there are a bunch of firmwares out there that "swallow" these initial `M110` handshake commands without acknowledgement and which also don't send a proper `start` response that could be used to sync things up again. In order to reliably get a communication going, there's a minimal asychronicityintroduced here to ensure maximum compatibility.

I'm not happy with that either, but I don't see any way around it that doesn't break things one way or the other for printers out there.

As I said, the issue I observed here was made more likely to happen by the `M25` behaviour of Klipper. It can still happen with a blocking `M25` though due to this asychronicity.

> The third issue is that after cancelling or pausing an sdcard print, OctoPrint continues to send an endless stream of M27 commands. (It does this even if Klipper responds with "Not SD printing.".) The log above shows this. I'm not sure if that is expected behaviour or not.

This is expected behaviour. The reason is that an endless stream of users kept complaining that if they start a print through their printer's display thingy, OctoPrint would not notice. So I jumped through some hoops to try to figure out if an SD print has been triggered externally. And since the only way to figure this out is to constantly ask the printer "Hey, are you printing?" once an SD file has been selected, because there's no trigger on the serial interface that goes like "print started" or something like that that I could rely on here. Hence the spamming of `M27`. Not happy about this either, but it's the best I could come up with the available commands and behaviour across printer firmware variants. 

Did I mention yet how much I hate the whole "protocol" situation? 

And for the record, I'm actually allowed to bitch about this situation because I *did* try to change it in the past but was met by so much resistance and outright aggressive hostility by the community that I had to give up for the sake of my mental wellbeing 🤷‍♀️

foosel, 2019-10-31:
> However, it seems that OctoPrint is accepting past acknowledgements for future commands - which doesn't seem correct.

If you have a good idea on how to sync up acknowledgements with commands on firmwares where the acknowledgements can come out-of-order compare to the sent commands (yep, those exist), no `start` is sent and no line numbers or anything else identifiable is attached to the acknowledgements either - I'm all ears. As things are, I simply don't see any solution that won't break things with a multitude of printers in the field, leave communication stuck at some point or the other or a multitude of other problems. 

I can't even reliably detect non offending firmware because in order to know that a firmware behaves correctly I first have to talk to it, so the handshake (where the issue can arise) has to have happened already.

And leaving something like this up to the user to configure from my experience is only going to end in a support nightmare.

So... yeah. I'm open for ideas. I don't have any left.

KevinOConnor, 2019-10-31:
Well, I was thinking:
```diff
--- a/src/octoprint/util/comm.py
+++ b/src/octoprint/util/comm.py
@@ -512,7 +512,7 @@ class MachineCom(object):
                self._pausing_commands = settings().get(["serial", "pausingCommands"])
                self._emergency_commands = settings().get(["serial", "emergencyCommands"])
 
-               self._clear_to_send = CountedEvent(name="comm.clear_to_send", minimum=None)
+               self._clear_to_send = CountedEvent(name="comm.clear_to_send", minimum=None, maximum=1)
                self._send_queue = SendQueue()
                self._temperature_timer = None
                self._sd_status_timer = None
```

I'm really quite puzzled with the idea of a firmware acknowledging a command that has not been sent.  I can't grasp what that would even mean?!  I would guess that a firmware doing that is incompatible with acknowledgements and one might as well just disable acks in that case.

I suppose an equivalent of the following code could be an alternative for Klipper:
```diff
--- a/src/octoprint/util/comm.py
+++ b/src/octoprint/util/comm.py
@@ -1941,6 +1941,7 @@ class MachineCom(object):
                                                elif "klipper" in firmware_name.lower():
                                                        self._logger.info("Detected Klipper firmware, enabling relevant features for issue free communication")
 
+                                                       self._clear_to_send._max = 1 # XXX - replace with equivalent method call
                                                        self._unknownCommandsNeedAck = True
 
                                                self._firmware_info_received = True
```

Puzzled,
-Kevin

foosel, 2019-10-31:
The second one looks like something I'm happy to add. I'll also look into making it a general setting that just gets applied for Klipper but can also be set manually if one knows their firmware and printer controller to be well behaved. 

The first one was in the code in the past and led to a ton of problems in the field.

The puzzledness switched to utter frustration for me some time around 2014.

KevinOConnor, 2019-11-01:
> The second one looks like something I'm happy to add.

Okay, thanks.  Unfortunately, it doesn't work.  After some furhter testing, OctoPrint rapidly gets out of sync with the acks.  For example:
```
2019-10-31 23:46:23,694 - Connecting to: /tmp/printer
2019-10-31 23:46:23,699 - Changing monitoring state from "Offline" to "Opening serial port"
2019-10-31 23:46:23,704 - Connected to: Serial<id=0x67908330, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, 
dsrdtr=False), starting monitor
2019-10-31 23:46:23,705 - Changing monitoring state from "Opening serial port" to "Connecting"
2019-10-31 23:46:23,721 - Send: N0 M110 N0*125
2019-10-31 23:46:23,723 - Recv: ok
2019-10-31 23:46:23,728 - Send: N0 M110 N0*125
2019-10-31 23:46:23,734 - Changing monitoring state from "Connecting" to "Operational"
2019-10-31 23:46:23,754 - Recv: ok
2019-10-31 23:46:23,758 - Send: N0 M110 N0*125
2019-10-31 23:46:23,759 - Recv: ok
2019-10-31 23:46:23,761 - Send: N1 M115*39
2019-10-31 23:46:23,763 - Recv: ok FIRMWARE_VERSION:v0.6.0-1621-g205e4ed FIRMWARE_NAME:Klipper
2019-10-31 23:46:23,770 - Send: M21
2019-10-31 23:46:23,772 - Recv: // Can not update MCU 'mcu' config as it is shutdown
2019-10-31 23:46:23,773 - Recv: // Once the underlying issue is corrected, use the
2019-10-31 23:46:23,774 - Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
2019-10-31 23:46:23,775 - Recv: // config, and restart the host software.
2019-10-31 23:46:23,776 - Recv: // Error configuring printer
2019-10-31 23:46:23,777 - Recv: !! Can not update MCU 'mcu' config as it is shutdown
2019-10-31 23:46:23,780 - Recv: ok
2019-10-31 23:46:28,025 - Send: firmware_restart
2019-10-31 23:46:28,040 - Recv: ok
2019-10-31 23:46:28,336 - Recv: // Klipper state: Disconnect
2019-10-31 23:46:28,735 - Send: M105
2019-10-31 23:46:29,567 - Recv: ok T:0
2019-10-31 23:46:31,587 - Recv: // Klipper state: Ready
2019-10-31 23:46:33,736 - Send: M105
2019-10-31 23:46:33,743 - Recv: ok B:25.3 /0.0 T0:24.8 /0.0
2019-10-31 23:46:34,624 - Send: G28
2019-10-31 23:46:35,749 - Recv: ok
2019-10-31 23:46:38,712 - Send: G91
2019-10-31 23:46:38,729 - Recv: ok
2019-10-31 23:46:38,734 - Send: G1 Z-10 F6000
2019-10-31 23:46:38,740 - Recv: ok
2019-10-31 23:46:38,746 - Send: G90
2019-10-31 23:46:38,750 - Recv: ok
2019-10-31 23:46:38,753 - Send: M105
2019-10-31 23:46:38,757 - Recv: ok B:25.5 /0.0 T0:25.0 /0.0
2019-10-31 23:46:40,627 - Send: G91
2019-10-31 23:46:40,638 - Recv: ok
2019-10-31 23:46:40,645 - Send: G1 Z-10 F6000
2019-10-31 23:46:40,648 - Recv: ok
2019-10-31 23:47:10,671 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
```

I believe the `self._clear_to_send.clear()` in both `_send_loop()` and `_emergency_force_send()` would need to be done before `self._do_send()`.  Otherwise, it would seem that the 'ok' response can be received before the clear() is issued and OctoPrint can get out of sync.

I'm testing with that additional change now.

-Kevin

foosel, 2019-11-01:
> I believe the `self._clear_to_send.clear()` in both `_send_loop()` and `_emergency_force_send()` would need to be done before `self._do_send()`. Otherwise, it would seem that the 'ok' response can be received before the clear() is issued and OctoPrint can get out of sync.

Oh, that is a good point!

foosel, 2019-11-01:
After some first tests, including some induced timeouts and firmware quirks, this looks actually fairly promising:

``` patch
diff --git a/src/octoprint/util/comm.py b/src/octoprint/util/comm.py
index 4c4991fd7..517b9fc7c 100644
--- a/src/octoprint/util/comm.py
+++ b/src/octoprint/util/comm.py
@@ -3186,9 +3186,26 @@ class MachineCom(object):
 					# at hand here and only clear our clear_to_send flag later if that's the case
 					gcode, subcode = gcode_and_subcode_for_cmd(command)
 
+					def use_up_a_clear():
+						# we only need to use up a clear if the command we just sent was either a gcode command or if we also
+						# require ack's for unknown commands
+						use_up_clear = self._unknownCommandsNeedAck
+						if gcode is not None:
+							use_up_clear = True
+
+						if use_up_clear:
+							# if we need to use up a clear, do that now
+							self._clear_to_send.clear()
+						else:
+							# Otherwise we need to tickle the read queue - there might not be a reply
+							# to this command, so our _monitor loop will stay waiting until timeout. We
+							# definitely do not want that, so we tickle the queue manually here
+							self._continue_sending()
+
 					if linenumber is not None:
 						# line number predetermined - this only happens for resends, so we'll use the number and
 						# send directly without any processing (since that already took place on the first sending!)
+						use_up_a_clear()
 						self._do_send_with_checksum(command, linenumber)
 
 					else:
@@ -3239,6 +3256,7 @@ class MachineCom(object):
 							continue
 
 						# now comes the part where we increase line numbers and send stuff - no turning back now
+						use_up_a_clear()
 						self._do_send(command, gcode=gcode)
 
 					# trigger "sent" phase and use up one "ok"
@@ -3247,21 +3265,6 @@ class MachineCom(object):
 						on_sent()
 					self._process_command_phase("sent", command, command_type, gcode=gcode, subcode=subcode, tags=tags)
 
-					# we only need to use up a clear if the command we just sent was either a gcode command or if we also
-					# require ack's for unknown commands
-					use_up_clear = self._unknownCommandsNeedAck
-					if gcode is not None:
-						use_up_clear = True
-
-					if use_up_clear:
-						# if we need to use up a clear, do that now
-						self._clear_to_send.clear()
-					else:
-						# Otherwise we need to tickle the read queue - there might not be a reply
-						# to this command, so our _monitor loop will stay waiting until timeout. We
-						# definitely do not want that, so we tickle the queue manually here
-						self._continue_sending()
-
 				finally:
 					# no matter _how_ we exit this block, we signal that we
 					# are done processing the last fetched queue entry
@@ -3788,10 +3791,10 @@ class MachineCom(object):
 			return
 
 		self._logger.info(message)
-		self._do_send(cmd, gcode=gcode)
 
 		# use up an ok since we will get one back for this command and don't want to get out of sync
 		self._clear_to_send.clear()
+		self._do_send(cmd, gcode=gcode)
 
 		return None,
 
```

foosel, 2019-11-01:
Further adjustment (`ackMax` defaulting to 1...):

``` patch
diff --git a/src/octoprint/util/comm.py b/src/octoprint/util/comm.py
index 4c4991fd7..80196735c 100644
--- a/src/octoprint/util/comm.py
+++ b/src/octoprint/util/comm.py
@@ -512,7 +512,8 @@ class MachineCom(object):
 		self._pausing_commands = settings().get(["serial", "pausingCommands"])
 		self._emergency_commands = settings().get(["serial", "emergencyCommands"])
 
-		self._clear_to_send = CountedEvent(name="comm.clear_to_send", minimum=None)
+		self._ack_max = settings().getInt(["serial", "ackMax"])
+		self._clear_to_send = CountedEvent(name="comm.clear_to_send", minimum=None, maximum=self._ack_max)
 		self._send_queue = SendQueue()
 		self._temperature_timer = None
 		self._sd_status_timer = None
@@ -3143,6 +3144,19 @@ class MachineCom(object):
 			self._logger.debug("Type already in send queue: " + e.type)
 			return False
 
+	def _use_up_clear(self, gcode):
+		# we only need to use up a clear if the command we just sent was either a gcode command or if we also
+		# require ack's for unknown commands
+		eats_clear = self._unknownCommandsNeedAck
+		if gcode is not None:
+			eats_clear = True
+
+		if eats_clear:
+			# if we need to use up a clear, do that now
+			self._clear_to_send.clear()
+
+		return eats_clear
+
 	def _send_loop(self):
 		"""
 		The send loop is responsible of sending commands in ``self._send_queue`` over the line, if it is cleared for
@@ -3189,6 +3203,7 @@ class MachineCom(object):
 					if linenumber is not None:
 						# line number predetermined - this only happens for resends, so we'll use the number and
 						# send directly without any processing (since that already took place on the first sending!)
+						self._use_up_clear(gcode)
 						self._do_send_with_checksum(command, linenumber)
 
 					else:
@@ -3239,7 +3254,13 @@ class MachineCom(object):
 							continue
 
 						# now comes the part where we increase line numbers and send stuff - no turning back now
+						used_up_clear = self._use_up_clear(gcode)
 						self._do_send(command, gcode=gcode)
+						if not used_up_clear:
+							# If we didn't use up a clear we need to tickle the read queue - there might
+							# not be a reply to this command, so our _monitor loop will stay waiting until
+							# timeout. We definitely do not want that, so we tickle the queue manually here
+							self._continue_sending()
 
 					# trigger "sent" phase and use up one "ok"
 					if on_sent is not None and callable(on_sent):
@@ -3247,21 +3268,6 @@ class MachineCom(object):
 						on_sent()
 					self._process_command_phase("sent", command, command_type, gcode=gcode, subcode=subcode, tags=tags)
 
-					# we only need to use up a clear if the command we just sent was either a gcode command or if we also
-					# require ack's for unknown commands
-					use_up_clear = self._unknownCommandsNeedAck
-					if gcode is not None:
-						use_up_clear = True
-
-					if use_up_clear:
-						# if we need to use up a clear, do that now
-						self._clear_to_send.clear()
-					else:
-						# Otherwise we need to tickle the read queue - there might not be a reply
-						# to this command, so our _monitor loop will stay waiting until timeout. We
-						# definitely do not want that, so we tickle the queue manually here
-						self._continue_sending()
-
 				finally:
 					# no matter _how_ we exit this block, we signal that we
 					# are done processing the last fetched queue entry
@@ -3788,10 +3794,12 @@ class MachineCom(object):
 			return
 
 		self._logger.info(message)
-		self._do_send(cmd, gcode=gcode)
 
 		# use up an ok since we will get one back for this command and don't want to get out of sync
-		self._clear_to_send.clear()
+		used_up_clear = self._use_up_clear(gcode)
+		self._do_send(cmd, gcode=gcode)
+		if not used_up_clear:
+			self._continue_sending()
 
 		return None,
 
```

If that doesn't blow up into my face during further testing also with out-of-band commands you'll have done this project a huge service here @KevinOConnor. Though I'm still very skeptical about the possible side effects here, the thought of doing it this way around simply never occurred to me and a set of fresh eyes here is *ultra* useful in any case!

KevinOConnor, 2019-11-02:
Thanks!  I'm hoping it works out in the general case.

FYI, I haven't noticed any issues locally since I moved the clear() above the do_send().

-Kevin

foosel, 2019-11-05:
Pushed that now as 2263ebde72a864792f0c9fb0cb10f0e731438e64

commit 1d602509 (2019-10-30): 🐛 Only detect sd printing on byte pos change

Should fix #3301, or at least make it way less likely to happen.

The problem seems to have been made worse by `M25` not being blocking
on Klipper and thus the SD printing not being guaranteed to have stopped
yet after its return, meaning the following `M26 S0` will be denied,
causing the file position to not be properly reset and hence the issue
to arise in the first place.
closed 2020-03-04 by foosel