OctoPrint/OctoPrint #3302: Timelapse settings are reverted after restart

sabag, 2019-10-10:
I have OctoPrint 1.3.11 installed on Raspberry pi 3.
I had Timelapse on for several weeks, now whenever I disable it (change to "off"), and restart the Pi, the settings are reverted to "on z change" (like it was).

steps to reproduce (i think):
1. enable timelapse
2.leave it on for several weeks, many many restarts in between
3.disable timelapse (change to "off")
4.restart pi
--> timelapse settings are reverted to "On Z Change"

attached a screen recording of the issue

[octoprint - timelapse config reverted after restart.zip](https://github.com/foosel/OctoPrint/files/3712299/octoprint.-.timelapse.config.reverted.after.restart.zip)

foosel, 2019-10-31:
Does the same happen when you persist "off" as the new **default**? You didn't do that here, and if you don't save it as default it will only apply to the current session so to speak. If you want to persist it across restarts you have to set a default.

sabag, 2019-10-31:
When I switch to "off" the "save as default" is disabled :

[image]

foosel, 2019-10-31:
Can you try connecting to the printer first?

sabag, 2019-10-31:
you right, with the printer connected it is enabled. thanks for fixing..

foosel, 2019-11-05:
Closing this since the timelapse settings in fact do not revert when they are told to persist, and the issue with the save box not being clickable unless connected to a printer has been resolved by the above commit.

closed 2019-11-05 by foosel