OctoPrint/OctoPrint #3304: [BUG] Octoprint Reporting PIP not installed when version > 19.0.3

rwmech, 2019-10-11:
#### What were you doing?

Octoprint reported PIP was not installed.  Also restoring a backup from another instance reported an error.  octoprint.log indicated "Unknown Pip".

Steps to reproduce
1. Change to a version of PIP higher than 19.0.3.    bin/pip install pip==19.2
2. Restart Octoprint
3. Go to plugins, attempt to install.  Reports PIP is not available. 
4. Change to a version of PIP equal to or below 19.0.3 bin/pip install pip==19.0.3
5. Restart Octoprint
6. Go to plugins, all works well.

#### What did you expect to happen?
Since documentation says minimum version is 19.0.3 for Octoprint 1.3.11 it should work normally

#### What happened instead?
Octoprint reported that PIP was not installed.  It would also not restore backups on a fresh install. 

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint
1.3.11

#### Operating System running OctoPrint
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.3 LTS
Release:        18.04
Codename:       bionic

#### Printer model & used firmware incl. version
N/A

#### Browser and version of browser, operating system running browser
N/A

#### Link to octoprint.log
https://pastebin.com/L5AuuJfT

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-10-13:
It's not the version of pip that's the problem, OctoPrint can't use your setup to install stuff because the target location isn't writable by it:

```
 2019-10-11 15:59:18,505 - octoprint.util.pip - ERROR - Cannot use this pip install, can't write to the install dir and also can't use --user for installing. Check your setup and the permissions on /home/octo3/OctoPrint/venv/lib/python2.7/site-packages.
```

You might want to check what's up there. Feel free to seek support on the forums with this.

closed 2019-10-13 by foosel

rwmech, 2019-10-13:
I had fixed that problem and still encountered the issue with unknown pip.  While a separate issue I still think you should try replicating the issue.  The permissions issue I resolved prior.

foosel, 2019-10-14:
Hm, all I can see in the log is permission errors, are you sure you fixed it? If so can you provide a log with a reproduction with fixed permissions? I'll also try to reproduce ASAP

foosel, 2019-10-14:
I cannot reproduce this:

```
2019-10-14 14:18:04,864 - octoprint.startup - INFO - ******************************************************************************
2019-10-14 14:18:04,867 - octoprint.startup - INFO - Starting OctoPrint 1.3.11
2019-10-14 14:18:04,867 - octoprint.startup - INFO - ******************************************************************************
2019-10-14 14:18:06,064 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2019-10-14 14:18:06,070 - octoprint.startup - INFO - Blacklist processing done, adding 1 blacklisted plugin versions: roomtemp (any)
2019-10-14 14:18:06,110 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-10-14 14:18:08,302 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 18 mixin implementations, 25 hook handlers
2019-10-14 14:18:08,600 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2019-10-14 14:18:08,629 - octoprint.server - INFO - Intermediary server started
2019-10-14 14:18:08,630 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2019-10-14 14:18:09,398 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 18 mixin implementations, 25 hook handlers
2019-10-14 14:18:09,441 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2019-10-14 14:18:09,443 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2019-10-14 14:18:09,542 - octoprint.plugins.softwareupdate - INFO - Version cache was created for another version of OctoPrint, not using it
2019-10-14 14:18:11,824 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2019-10-14 14:18:14,000 - octoprint.util.pip - INFO - Version of pip is 19.2.3
2019-10-14 14:18:14,001 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-10-14 14:18:14,003 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-10-14 14:18:14,008 - octoprint.plugin.core - INFO - Initialized 18 plugin implementation(s)
2019-10-14 14:18:14,023 - octoprint.plugin.core - INFO - 18 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Github release patcher (0.1.0) = /home/pi/.octoprint/plugins/github_release_patcher.py
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
|  Webcam Tab (0.1.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_webcamtab
|  Yamlpatcher (0.1.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_yamlpatcher
2019-10-14 14:18:14,043 - octoprint.environment - INFO - Detected environment is Python 2.7.13 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1200.0
|    ram: 918192128
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Rev 1.2
|      octopi_version: 0.15.1
|  python:
|    pip: 19.2.3
|    version: 2.7.13
|    virtualenv: /home/pi/oprint
2019-10-14 14:18:14,052 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/webassets...
2019-10-14 14:18:14,054 - octoprint.server - INFO - Reset webasset folder /home/pi/.octoprint/generated/.webassets-cache...
2019-10-14 14:18:15,106 - octoprint.server - INFO - Shutting down intermediary server...
2019-10-14 14:18:15,577 - octoprint.server - INFO - Intermediary server shut down
2019-10-14 14:18:15,580 - octoprint.events - INFO - Processing startup event, this is our first event
2019-10-14 14:18:15,581 - octoprint.events - INFO - Adding 1 events to queue that were held back before startup event
[... cut ...]
2019-10-14 14:18:19,126 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2019-10-14 14:18:19,128 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2019-10-14 14:18:19,129 - octoprint.util.pip - INFO - ==> pip ok -> yes
```

[image]

Just for good measure I also tried with 1.3.12rc3 and a locally run instance of 1.3.13.dev. No issues at all with pip versions larger than 19.0.3.

rwmech, 2019-10-14:
You are running this under a PI and not under a venv so that could be the
difference.  I'll run this later tonight and provide a log.

foosel, 2019-10-14:
`/home/pi/oprint` on OctoPi *is* a venv ;) I've run it in that twice with varying version combinations of OctoPrint and pip, and also on a local checkout in a local venv.