OctoPrint/OctoPrint #3310: watched folder does not always work

schnello, 2019-10-21:
## What did you expect to happen?
All files are uploaded

#### What happened instead?
Sometimes the files are ignored (2-3 per week if i upload 3 Files per day)

#### Version of OctoPrint
1.3.12 RC3

#### Operating System running OctoPrint
Ubuntu 18 Server

#### Printer model & used firmware incl. version
I3 Mega with Klipper FW

#### Browser and version of browser, operating system running browser
Chrome

#### Link to octoprint.log

Hello.

Sometime the files in the watched folder are ignored.
Example1:
```
2019-10-19 22:06:38,287 - octoprint.server.util.watchdog - DEBUG - File at /home/nas/.octoprint/watched/Laser_Ring.gcode is no longer growing, counting down: 5
2019-10-19 22:06:38,298 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-10-19 22:06:38,307 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-10-19 22:06:38,363 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-10-19 22:06:38,639 - octoprint.events.fire - DEBUG - Firing event: ZChange (Payload: {'new': 1.652, 'old': 1.552})
2019-10-19 22:06:38,640 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2019-10-19 22:06:38,643 - octoprint.plugin - DEBUG - Calling on_event on announcements
2019-10-19 22:06:38,644 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2019-10-19 22:06:38,644 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check
2019-10-19 22:06:38,645 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2019-10-19 22:06:38,645 - octoprint.plugin - DEBUG - Calling on_event on tracking
2019-10-19 22:06:38,645 - octoprint.plugin - DEBUG - Calling on_event on DryRun
2019-10-19 22:06:38,646 - octoprint.plugin - DEBUG - Calling on_event on PrintTimeGenius
2019-10-19 22:06:38,646 - octoprint.plugin - DEBUG - Calling on_event on autoselect
2019-10-19 22:06:38,646 - octoprint.plugin - DEBUG - Calling on_event on excluderegion
2019-10-19 22:06:38,646 - octoprint.plugin - DEBUG - Calling on_event on filamentmanager
2019-10-19 22:06:38,646 - octoprint.plugin - DEBUG - Calling on_event on firmwareupdater
2019-10-19 22:06:38,646 - octoprint.plugin - DEBUG - Calling on_event on klipper
2019-10-19 22:06:38,647 - octoprint.plugin - DEBUG - Calling on_event on m73progress
2019-10-19 22:06:38,647 - octoprint.plugin - DEBUG - Calling on_event on pushover
2019-10-19 22:06:38,647 - octoprint.plugins.pushover - INFO - Got an event: ZChange Payload: {'new': 1.652, 'old': 1.552}
2019-10-19 22:06:38,648 - octoprint.plugins.pushover - INFO - Event triggered: ZChange 
2019-10-19 22:06:38,655 - octoprint.events.fire - DEBUG - Firing event: ZChange (Payload: {'new': 1.552, 'old': 1.652})
2019-10-19 22:06:38,656 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2019-10-19 22:06:38,657 - octoprint.plugin - DEBUG - Calling on_event on announcements
2019-10-19 22:06:38,658 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2019-10-19 22:06:38,659 - octoprint.plugin - DEBUG - Calling on_event on printer_safety_check

```

In this case the watchdog stops counting down.

```

If i move the file out of the folder and move it back.. it works as expected.
2019-10-19 22:06:39,139 - octoprint.plugin - DEBUG - Calling on_event on firmwareupdater
2019-10-19 22:06:39,141 - octoprint.plugin - DEBUG - Calling on_event on klipper
2019-10-19 22:06:39,143 - octoprint.plugin - DEBUG - Calling on_event on m73progress
2019-10-19 22:06:39,143 - octoprint.plugin - DEBUG - Calling on_event on pushover
2019-10-19 22:06:39,143 - octoprint.plugins.pushover - INFO - Got an event: ZChange Payload: {'new': 1.712, 'old': 1.552}
2019-10-19 22:06:39,145 - octoprint.plugins.pushover - INFO - Event triggered: ZChange 
2019-10-19 22:06:39,288 - octoprint.server.util.watchdog - DEBUG - File at /home/nas/.octoprint/watched/Laser_Ring.gcode is no longer growing, counting down: 4
2019-10-19 22:06:40,200 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-10-19 22:06:40,215 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-10-19 22:06:40,289 - octoprint.server.util.watchdog - DEBUG - File at /home/nas/.octoprint/watched/Laser_Ring.gcode is no longer growing, counting down: 3
2019-10-19 22:06:41,291 - octoprint.server.util.watchdog - DEBUG - File at /home/nas/.octoprint/watched/Laser_Ring.gcode is no longer growing, counting down: 2
2019-10-19 22:06:41,366 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-10-19 22:06:42,190 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-10-19 22:06:42,202 - octoprint.server.util.flask - DEBUG - App sessions after cleanup: {}
2019-10-19 22:06:42,292 - octoprint.server.util.watchdog - DEBUG - File at /home/nas/.octoprint/watched/Laser_Ring.gcode is no longer growing, counting down: 1
2019-10-19 22:06:42,292 - octoprint.server.util.watchdog - DEBUG - File at /home/nas/.octoprint/watched/Laser_Ring.gcode is stable, moving it
2019-10-19 22:06:42,360 - octoprint.filemanager.analysis - DEBUG - Adding entry local:Laser_Ring.gcode to analysis queue with high priority
2019-10-19 22:06:42,360 - octoprint.filemanager.analysis - DEBUG - Processing entry local:Laser_Ring.gcode from queue (priority 50)
2019-10-19 22:06:42,361 - octoprint.events.fire - DEBUG - Firing event: FileAdded (Payload: {'path': u'Laser_Ring.gcode', 'storage': 'local', 'name': u'Laser_Ring.gcode', 'type': ['machinecode', 'gcode']})
2019-10-19 22:06:42,362 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2019-10-19 22:06:42,362 - octoprint.plugin - DEBUG - Calling on_event on announcements
2019-10-19 22:06:42,363 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
```

[octoprint.log](https://github.com/foosel/OctoPrint/files/3749658/octoprint.log)

In other cases i found no "watchdog" information in the log. Looks like the watchdog was not started.

schnello, 2019-10-21:
Addional comment:
I know one of the main questions are: Have you the same issue if you run it in "safe mode"
I try to answer this question with an automatic script. 
Copy file to watched folder, count until 10, look is the file uploaded and if yes... delete it.
But this needs some time.
[image]

schnello, 2019-10-22:
Ok. After 1231 copy, upload and delete actions i can say:
The watchdog works:
perfect with the local file system (cp, mv, etc)
perfect with an CIFS share
unstable with DAVFS (https://wiki.ubuntuusers.de/WebDAV/)

I guess this is because the DAVFS System generates as first action only the .part and rename the file to the final file name. I dont know why it works for the first 20-100 trys.... but after Nr100 more and more often it becomes a problem.

From my view... the case could be closed. DAVFS is not a standard file system. But good to know that the watchdog has problems with this kind of storage.

foosel, 2019-10-30:
Sorry, I was out with the flu for a while and that fell through. Looks indeed like a limitation of the used watchdog implementation. I'm not too keen on trying to work around that to be honest since it seems a bit too much like a rare edge case. So I'm closing this. If anyone wants to tackle this however, I'd be happy to accept a PR.