OctoPrint/OctoPrint #3313: Officially migrate to Python 3

chudsaviet, 2019-10-22:
**Is your feature request related to a problem? Please describe.**
Python 2.7 will not be maintained past 2020.

**Describe the solution you'd like**
Migrate to Python 3.

**Describe alternatives you've considered**
I don't see any alternatives.

**Additional context**
https://pythonclock.org/

tedder, 2019-10-22:
This has been "officially" in progress. It's a well-known thing. It's talked about in every Octoprint On Air and in the discussion forums. Work has been done for a while to get there. I think the master ticket is #2722.

foosel, 2019-10-23:
Yep. Duplicate. I'm aware and have been talking about almost nothing else the past months or even year. Will be in 1.4.0, trying to get that out as fast as possible this year still, might end up just being in RC stage because of health reasons. Help testing `devel` and the upcoming devel RCs if you want to see this happening.

closed 2019-10-23 by foosel

chudsaviet, 2019-10-23:
@foosel , sorry, looks like have not searched enough.