OctoPrint/OctoPrint #3316: [Request] Supply a "Job UUID" to plugin in the return value of `get_current_job`

kennethjiang, 2019-10-24:
**Is your feature request related to a problem? Please describe.**

[My plugin](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective) needs the ability to correlate individual events by the print job. For instance, I'd like to know a PrintFailed event is for the same print job as the PrintCancelled event at some earlier point, and the PrintResumed event at another point earlier.

Since I can't find a mechanism in existing APIs to accomplish this, I had to craft a way to track Jobs and assign UUID inside the plugin, by reverse-engineer the print job's lifecycle.

However, users have reported mis-identified print jobs. I suspect that the problem is caused by either

- some events like `PrintStarted` occasionally fail to be delivered to my plugin, or

- there is a bug deep inside my code to track print job life-cycle and hence attaches the wrong UUID.

The problem is not reproducible, and hence very difficult to debug.

**Describe the solution you'd like**

Since there must already be an Job object somewhere in OctoPrint, I guess it'd be a lot easier and more robust to add a UUID to that object and expose it to plugins via `get_current_job` method call.

**Describe alternatives you've considered**

As stated earlier in this issue, I have implemented an alternative, which turns out to be buggy and hard to debug/maintain.

Please let me know if I missed something in existing OctoPrint APIs that can provide such mechanism or help me achieve the same goal.

**Additional context**

I have also [brought this up in OctoPrint forum](https://community.octoprint.org/t/identify-a-print-job/12942/9).

OllisGit, 2019-10-26:
hmmm...why is it not possible to listen to "PRINT_STARTED"-Event and set a local variable in your plugin e.g. job_id = currentTimeIncludingSeconds + filename
Each following event is (of course) related to that job_id.

I did this in my plugins all the time or do you have several octoprint-servers and you need a unique id over all jobs.

I also tried to use the python command ```id(self._printer.get_current_job())```, but each time there is a new object id.

Sorry, didn't get it...is your plugin opensource so that is possible to do a code-review.

BR
Olli

jneilliii, 2019-10-26:
It's the spaghetti detective plugin. He is currently doing the print start event method, but it's not reliable based on user feedback.

kennethjiang, 2019-10-26:
@OllisGit Thank you for offering to review the code.

The plugin code is currently do basically what you said: https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/blob/master/octoprint_thespaghettidetective_beta/print_event.py#L14

The server code is also open source but the code to process this unique id is a lot more complicated. Will be happy to point you to that if you are interested.

OllisGit, 2019-10-26:
Hi @kennethjiang,
I looked into your octoprint print code  and documented my findings in the [forums thread](https://community.octoprint.org/t/identify-a-print-job/12942/13), because this issue is a feature request of an uuid.

BR
Olli