OctoPrint/OctoPrint #3327: [Request] Do not return code 401 from 'api/login' if user/pass is wrong but rather 403

anthonyst91, 2019-11-04:
Hello Gina, 

Working on Printoid, I face an issue with the 'api/login' API.
Especially when the parameter "pass" or "user" query parameter is wrong (because the user entered a wrong credential value)

API documentation found here:
[http://docs.octoprint.org/en/master/api/general.html#post--api-login](http://docs.octoprint.org/en/master/api/general.html#post--api-login
)

#### What were you doing?

Calling the following URL from the Printoid application (on Android):
_https://server_url/api/login?user=XXX&pass=YYY&remember=true&passive=false_

Using the OkHttp library as HTTP client, built using the "authenticator" callback (used for reactive authentication) : [https://square.github.io/okhttp/4.x/okhttp/okhttp3/-authenticator/](https://square.github.io/okhttp/4.x/okhttp/okhttp3/-authenticator/)

And passing wrong credentials (XXX and YYY in call)

#### What did you expect to happen?

Receiving a response with error code 403 (forbidden)
[https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/403](https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/403)

#### What happened instead?

Receiving a response with error code 401 (unauthorized)
[https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/401
](https://developer.mozilla.org/en-US/docs/Web/HTTP/Status/401)

In fact, we cannot receive a code 401 in this case because it tells our HTTP client that the basic authentication has failed, so the 'authenticate' callback of our OkHttp client is called again and again, until we stop to send the request (generally after 3 retries)

Unfortunately this is not the expected behavior, because in this case our client is warned that the basic authentication is wrong, whereas it is fine. A 403 would be better because it would say that "the server understood the request but refuses to authorize it", which is more compliant with the current case.

In fact, the 'login' API already returns a 403 in case of the requested user account has been removed - may you could return a 403 for both cases, but with a different message in the body (such as "username or password is wrong" and "the account does not exist anymore")

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

v1.3.12 but started to occur since the integration of the 'login' API

#### Operating System running OctoPrint

OctoPi v0.16.0

foosel, 2019-11-05:
Looks like that was missed when fixing #2297 🤨

Fixed by the above commit, will be in the next release.

anthonyst91, 2019-11-05:
Damn, I did a search on opened tickets but I missed to search on already closed tickets - my bad for that. 

Thanks for your quick feedback :) Today I managed to treat the "wrong credentials" case by catching 401 error + body message, but it would be indeed more comfortable with a 403. Waiting for the next release then!

Keep up the great work!

closed 2020-03-04 by foosel

linehammer, 2020-11-09:
This error indicates that the server has determined that you are not allowed access to the thing you've requested, either on purpose or due to a misconfiguration . It's probably because the site owner has limited access to it and you don't have permission to view it.  The vast majority of the time, there's not much you can do to fix things on your (*client) end. There are three common causes for [403 Forbidden](http://net-informations.com/q/mis/403.html) error (server side) . Here they are listed from most likely to least likely:

- An empty website directory
- No index page
- Incorrect settings in the .htaccess file
- Permission / Ownership error

If authentication credentials were provided in the request, the server considers them insufficient to grant access. The client SHOULD NOT automatically repeat the request with the same credentials. The client MAY repeat the request with new or different credentials. However, a request might be forbidden for reasons unrelated to the credentials.

commit 9cd597a4 (2019-11-05): 🐛 Fix invalid use of 401 response code

Leftovers from fixing #2297.

Fixes #3327