OctoPrint/OctoPrint #3329: Serial is being disconected (after latest ugrade)

leoheck, 2019-11-06:
I am using Octoprint version `1.3.12` with an `Ender 5` with `Marlin 1.1.9` firmware and the BLTouch support.

After temperatures get right, the serial is being disconnected.
I tested different gcodes, but the serial behaves the same.

```
State: Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2793)
```

The error on the logs

```
2019-11-06 12:11:39,123 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/home/pi/.local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2793, in _readline
    ret = self._serial.readline()
  File "/home/pi/.local/lib/python2.7/site-packages/octoprint/util/comm.py", line 4932, in readline
    c = self.read(1)
  File "/usr/local/lib/python2.7/dist-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2019-11-06 12:11:39,145 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
``` 

The full log is (here)[https://pastebin.com/cNrNjvks]

foosel, 2019-11-06:
For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org) (as [you've already done](https://community.octoprint.org/t/disconnection-issues-with-rpi-and-ender-3-pro/13177/5?u=foosel)). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2019-11-06 by foosel

leoheck, 2019-11-06:
It is not a question, it is a bug. Did you read it?

foosel, 2019-11-06:
Yes, I did read it, and that particular error message has an endless FAQ entry on why that is not something under OctoPrint's control but happens a layer beneath it and by what it is usually caused which was linked in the log itself. That it occurred in the vicinity of an update might be due to a change in power consumption, or simply be coincidence.