OctoPrint/OctoPrint #3331: [Request] ask to confirm if you delete a job from octoprint

LMS0815, 2019-11-07:
Please add a confirmation request if you hit the delete button of a print job in OctoPrint navbar.
This would ensure, that yo do not expediently delete jobs you actually intended to print.

To ally backwards compatibility you could enable/disable this behavior in "functions".
Same behavior for _Confirm delete print job_ as you already have for _Confirm start print job_.

reloxx13, 2019-11-15:
#vote against

no time to confirm while mass deleting/sorting

would annoy more as it helps.

LMS0815, 2019-11-19:
I am sure some people might find this option annoying. 
Therefor I requested an option which can be configured to the liking of the amdin.

medisoft, 2020-12-29:
I join to this request. :D

LazeMSS, 2021-03-31:
Available as plugin:
https://plugins.octoprint.org/plugins/inlineconfirm/