OctoPrint/OctoPrint #3334: G2 and G3 arcs not displaying properly when gcode is displayed

ThyImortalGamer, 2019-11-08:
#### What were you doing?

Uploaded a gcode file that contained g2 and g3 instructions.
Checked GCode viewer while printing for essentially a sanity check

#### What did you expect to happen?
It was expected that the arcs created by the g2 and g3 commands would display properly

#### What happened instead?
It appears the G2 and G3 commands are being parsed incorrectly, causing the g2 and g3 commands to be swapped. G2 created arcs should display and print in a clockwise direction, G3 created arcs should display and print in a counter-clockwise direction. Instead it was displayed in the opposite way.

(Model printed without issue)

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

Octoprint 1.3.12

#### Operating System running OctoPrint

Octopi 0.16.0

#### Printer model & used firmware incl. version

I don't believe this is related but just in case, Anet A8 running Marlin version Bugfix 1.1.x

#### Browser and version of browser, operating system running browser

Mozilla Firefox running on Parrot OS (Linux)

#### Link to octoprint.log

https://hastebin.com/uliwakenuv.js

#### Link to contents of terminal tab or serial.log

Not a communicaton issue
#### Link to contents of Javascript console in the browser

https://hastebin.com/bizedetika.css

#### Screenshot(s)/video(s) showing the problem:

[image]
I have read the FAQ.

These are the files this was tested with for this issue:
[testFiles.zip](https://github.com/foosel/OctoPrint/files/3821403/testFiles.zip)

foosel, 2019-11-08:
> It appears the G2 and G3 commands are being parsed incorrectly, causing the g2 and g3 commands to be swapped.

You are right and I wish all bugs were that easy to fix ^^ Thank you for the full set of information needed for analysis AND even the solution :)

Fixed in the above commit, will be part of the next release.

ThyImortalGamer, 2019-11-08:
^~^ honestly I would have fixed it an  put up a PR, but I didn't have the free time to search for that specific code.. Besides, not sure how many people are actually affected by that given the fact that arcs seem to be a rare thing in 3D printing. Glad I could give you an easy task 😋

commit 8173f8d0 (2019-11-08): 🐛 gcode viewer: Fix swapped G2/G3

Closes #3334
closed 2020-03-04 by foosel