OctoPrint/OctoPrint #3339: GCode viewer and actual print are showing "missing layers.

UberEclectic, 2019-11-14:
#### What were you doing?

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

UberEclectic, 2019-11-14:
Ugg. My submission was reset when I logged into the OctoPrint host to get the uname -a version.
I'll have to start the whole report over.  :-/

UberEclectic, 2019-11-14:
I am happy to share the gcode, but it's quite large and I do not see how to include a compressed binary archive to either of the mentioned file-sharing sites.

foosel, 2019-11-14:
I fear I'll really need access to GCODE with which to reproduce this. Do you maybe have a smaller minimal example file in your collection that shows the same behaviour?

foosel, 2019-11-14:
> In the case of the actual print the Z-axis was incremented without laying any filament, so the next layer was printing "in thin air" and the whole print was utterly failed as a result.

Also just for the record, that is very interesting. OctoPrint really just takes the GCODE file and streams it line by line to the printer. If no filament was layed down, either there were no commands in the file to lay it down or the firmware misinterpreted something. The only modification stock OctoPrint does to GCODE files as it sends them is tack on a line number and checksum during transmission to be able to compensate for errors.

Please also share a full `serial.log` of a failing print so we can compare what's in the file with what goes over the line.

Also please try safe mode - the GCODE viewer issue and the printing issue happen at different parts of the codebase and the backend can *definitely* be influenced by plugins.

UberEclectic, 2019-11-14:
Hi, @foosel!  I wondered about that.  Really my assumption was that you are passing the gcode straight through, so I was surprised that printing via OctoPrint vs. gcode on SD card had a different result.  I had 5 failed attempts, which I at first blamed on finicky filament before I started carefully inspecting the gcode in OctoPrint's built-in viewer and saw the "emtpy" layer.

UberEclectic, 2019-11-14:
Looks like I can just drag the gcode file into a comment.  =D

If you check it out on gcode.ws and step through the layers (cursor up/down) in the GCode viewer tab (click the tab to have the line jump to the current layer), and watch what happens between layer 6 and 7 and from layer 8 back down to 7, you'll see that it jumps to line 44 for layer 7.  Very strange.

Using Repetier Host to step through the layers, while looking at the GCode is properly showing layer 7 in tact.  That matches what Cura shows in its layer viewer, where the GCode was generated.

[TAZWHE05MM_CDK125-FocuserDustCover v6.gcode.gz](https://github.com/foosel/OctoPrint/files/3846746/TAZWHE05MM_CDK125-FocuserDustCover.v6.gcode.gz)

UberEclectic, 2019-11-14:
Incidentally, I came across this forum post, which I can't help but suspect is related.
I have some printer setup code at the start of every job...

https://community.octoprint.org/t/octoprint-inserts-do-nothing-layer-2/9969/3

foosel, 2019-11-18:
So I just took a look into the file, and the empty layers here are caused by "sort layers" being enabled by default:

[image]

If you disable it so that the viewer doesn't sort by Z height but shows the layers in the order they appear in the file (which isn't default because it can confuse the heck out of people) you see that all those layers are part of the nozzle cleaning routine at the start of the print.

Note that the sorting *only* happens in the viewer on the frontend, strictly on client side. There's nothing done to the file as it gets streamed to the printer by OctoPrint, at least not in core OctoPrint. It could be a thirdparty plugin causing this however, so please try in safe mode. Also enable `serial.log` and if it continues to happen in safe mode share that please.

foosel, 2020-01-23:
@UberEclectic Any outcome for the above suggestions?

closed 2020-04-06 by foosel