OctoPrint/OctoPrint #3340: Print Starts before "Start G-Code"

TheSheepKing151, 2019-11-18:
## What were you doing?

Hello.
I am using PrusaSlicer to generate my G-Code. In PrusaSlicer i have set a Start G-Code, which does a little extrude+swipe to get the filament into the nozzle. When I print from sd-card everything works fine.

When I tried to print with Octoprint, I noticed, that the printer didn't do the Start G-Code at all.
Even in the G-Code viewer it shows the first layer at height 3mm. The second layer looks like my startup G-Code, and the third layer is the actual first layer.
I suspect that this is because my start G-Code is at height 0.35mm.

Start G-Code:
G28 ; home all axes
G1 Z3 F5000 ; lift nozzle
G92 E0;
M140 S[first_layer_bed_temperature]        ; Most of the way there, but next command begins immediately
M104 S[first_layer_temperature]     ; Close, but below the plastic level for PLA, next command begins immediately
M190 S[first_layer_bed_temperature]       ; The longest delay will be between the last command and this, blocking
M109 S[first_layer_temperature]      ; Should go quickly by comparison, blocking
G1 X1 Y20 Z0.35 F5000.0 ; move to start-line position
G1 X1 Y200.0 Z0.35 F1500.0 E15 ; draw 1st line
G1 X1.4 Y200.0 Z0.35 F5000.0 ; move to side a little
G1 X1.4 Y20 Z0.35 F1500.0 E30 ; draw 2nd line
G92 E0 ; reset extruder 
G1 E4 F300 ; Retract 4mm of Filament on Extruder

#### What did you expect to happen?
I expected the same to happen as when printing the same G-Code from SD-Card.

#### What happened instead?

As described earlier, the printer printed a real layer from 3mm first, then the start G-Code then the real first layer.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.3.12

#### Operating System running OctoPrint

OctoPi 0.16.0

#### Printer model & used firmware incl. version

Tronxy XY-2 Pro 
But not really important, as the Octoprint G-Code viewer says the same.

#### Browser and version of browser, operating system running browser

Really doesn't matter

#### Link to octoprint.log

https://pastebin.com/EShkJriu

#### Link to contents of terminal tab or serial.log

https://pastebin.com/aWCZV0i8

#### Link to contents of Javascript console in the browser

Doesn't matter

#### Screenshot(s)/video(s) showing the problem:

http://puu.sh/EG4u7/53dbb4a830.png
https://puu.sh/EG4uv/15195efd9c.png
https://puu.sh/EG4v6/03bd3b5a2c.png
https://puu.sh/EG4vu/92b1fdfa7e.png

I have read the FAQ.

foosel, 2019-11-19:
Please share a full GCODE file with which to reproduce this, and enable and share `serial.log` of a reproduction.

TheSheepKing151, 2019-11-19:
Here is the serial.log.
https://pastebin.com/cZmKAgU1
[tronxy_xy2_tensioner_wheel.zip](https://github.com/foosel/OctoPrint/files/3862904/tronxy_xy2_tensioner_wheel.zip)

I have attatched a sample G-Code file, but it happens with every model.

TheSheepKing151, 2019-11-29:
@foosel have you looked into the logs? Could this be because my start G-Code extrudes at z = 0.35 mm?

foosel, 2019-11-29:
I haven't yet had a chance due to 1.4.0rc1

foosel, 2019-11-29:
Before we continue here, on first look your printer has a broken firmware build that misuses `wait` ("I'm waiting for you, please send more commands") as `busy` ("I can't process new stuff right now, please hold"). Seems to be making its rounds in China 🙄

Install the two workaround plugins from [here](https://github.com/foosel/OctoPrint/issues/2309#issuecomment-358713557) and see if that maybe already fixes things.

closed 2020-06-18 by foosel