OctoPrint/OctoPrint #3341: REQUEST: CR-10 v2 Filament Detector Support

KenLucke, 2019-11-19:
The current version of OctoPrint (using v1.3.12) apparently does not support the built-in filament breakage/end detector that the CR-10 v2 comes with natively.  When printing from the SD card, the machine will pause and alert you when the filament runs out or breaks.  When printing from OctoPrint, it does not.  Either that, or I'm experiencing an Eye-Dee-Ten-Tee error and cannot find the function.

Can someone please add the support for this to the current version, or add a plugin that covers it?

foosel, 2019-11-19:
Some more information on *how* the firmware supposedly alerts the host on a filament runout would be nice, otherwise no one without access to that printer will be able to look into this or even tell you if it can be done ;)

Also: OctoPrint already supports a number of well defined action commands. Firmware just needs to use these to implement alerting. Sadly most firmware vendors don't seem to know or care about what they are doing and keep reinventing the wheel. I'm not going to implement vendor specifics into core OctoPrint (it would make things become completely unmanageable), so if that is once again what happened here, it'll be a case of plugin only.

KenLucke, 2019-11-19:
I'd love to be able to provide that information, but I have no idea where to find it.  Any suggestions?  Perhaps I can force a runout on a From-SD-Print and look at a log somewhere to see what was sent.

(And I agree completely about vendors instituting proprietary solutions for things that others have already invented, making everything more difficult.)

foosel, 2019-11-20:
Enabling `serial.log`, triggering the scenario and then sharing the resulting log here would already go a long way :)

KenLucke, 2019-11-20:
OK, I'd e happy to do that, if you can provide me with instructions on how to enable the log and where to find the file afterwards.  I used to code (everything from BASIC, FORTRAN, COBOL, Pascal, PERL, etc., right on down to "on the metal" machine code), but it's been a while, and I'm not familiar with the system yet, just the top layer.

foosel, 2019-11-21:
You can find all of this information in the FAQ:

https://community.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

LMS0815, 2019-11-21:
> Also: OctoPrint already supports a number of well defined action commands. Firmware just needs to use these to implement alerting.

What should a Marlin FW send as a response? I am happy to open a request at the developer page...

jneilliii, 2019-11-21:
You may want to take a look [here](https://community.octoprint.org/t/octoprint-windows-pause-octoprint-when-printer-sends-m600-command/13067/2).

LMS0815, 2019-11-21:
THX
seems that it is defined:
`#define ACTION_ON_FILAMENT_RUNOUT "filament_runout"`

jneilliii, 2019-11-21:
If that's the case, then maybe using the [Actions Command](https://plugins.octoprint.org/plugins/actioncommands/) plugin would allow you to monitor for that action `filament_runout` and act upon it. Just not sure what command you would use in the settings.

foosel, 2020-07-22:
Closing due to lack of response.

closed 2020-07-22 by foosel

developnoop, 2020-07-23:
So, the OP didn't come up with a log file, but i can. 
I also have the Creality CR 10 V2, so i can provide the log file below.
I triggered manual the filament sensor shortly before the pause command is manually clicked in octoprint.
Also i am not so firm with the commands, but at least i could not see one filament releated command.

foosel, 2020-07-24:
>  i could not see one filament releated command.

Neither can I. Which means it can't be done, at least not unless the printer's firmware gets modified accordingly, and then one could already modify it to send established action commands.