OctoPrint/OctoPrint #3345: [request]  GCODE Scripts to generate M117 message before/after execution

LMS0815, 2019-11-21:
GCODE Script M117 messages.

I am trying to monitor my GCODE Scripts by adding M117 messages like:
```
**M117 Pree print UBL...**
;OCTO1 ; enable BI webstream
M104 S200; Set Hotend Temperature
M140 S60; Set Bed Temperature
G28 0; Homing -  If the position is known then exit without homing
G29 L; Load the previously-activated mesh
G29 F 10.0 ; Set Fade Height for correction at 10.0 mm
G29 A; Activate the UBL System
M190 S60; Wait for Bed Temperature
G29 J; probe 3 points and tilt the mesh according to what it finds.
**M117 Pree print DONE**
```
Unfortunately the second M117 message is mor or less imediately displayed although `G29 J` was not processed.

Is there any method to generate M117 messages synchronized with the script/printer?
Is there any method to ask the printer (Marlin) to send a generic messages to OctoPrint?

If not. Please add the ability to generate M117 messages based GCODE Script execution (on event?!).

foosel, 2019-11-21:
Congratulations, you've just learned about the move planner buffer in firmware.

Commands that trigger moves will usually be acknowledged immediately and executed asynchronously. To wait until all moves are finished before proceeding you can use `M400`. There's no way to otherwise synchronize with the planner buffer that I'm aware of, and I'm not aware of any ways to trigger messages to the host on planner buffer events (like enqueuing some form of action command execution).

I'm closing this since this is not something OctoPrint can do anything about but rather a case of firmware and/or gcode.

closed 2019-11-21 by foosel

LMS0815, 2019-11-21:
I am not sure if OctoPrint can not do anything:
_Please add the ability to generate M117 messages before/after GCODE Script execution (on event?!)._
If this is not possible consider generating a message as soon OctoPrint starts sending the Uploaded *.gcode to the printer. This would - at least - add some transparency to the state OctoPrint is in.

foosel, 2019-11-21:
You can already do that through a plugin. For events even through event subscriptions. It's in the docs.