OctoPrint/OctoPrint #3347: [RC Feedback] Feedback on the 1.4.0rc1 Release Candidate

foosel, 2019-11-25:
Please provide *general* feedback on your experience with the 1.4.0rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Unrelated third party plugin issues

schnello, 2019-11-25:
Installed on my Ubuntu 18 Server System with Klipper has FW. >> Installation OK and System is running. Later i perform some prints.

devildant, 2019-11-25:
hello @foosel ,
install ok but pb with cura octoprint plugin, when i try to upload a gcode file from cura : 
```

2019-11-25 17:09:34,089 - tornado.application - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 584, in _read_fixed_body
    ret = delegate.data_received(body)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/httpserver.py", line 311, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/routing.py", line 248, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2087, in data_received
    return self.handler.data_received(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 271, in data_received
    self._process_multipart_data(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 316, in _process_multipart_data
    self._on_request_body_finish()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 468, in _on_request_body_finish
    self._new_body += octoprint.util.to_bytes(p) + b'\r\n'
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
2019-11-25 17:09:34,346 - tornado.application - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 584, in _read_fixed_body
    ret = delegate.data_received(body)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/httpserver.py", line 311, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/routing.py", line 248, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2087, in data_received
    return self.handler.data_received(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 271, in data_received
    self._process_multipart_data(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 316, in _process_multipart_data
    self._on_request_body_finish()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 468, in _on_request_body_finish
    self._new_body += octoprint.util.to_bytes(p) + b'\r\n'
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
2019-11-25 17:11:29,371 - tornado.application - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 584, in _read_fixed_body
    ret = delegate.data_received(body)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/httpserver.py", line 311, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/routing.py", line 248, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2087, in data_received
    return self.handler.data_received(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 271, in data_received
    self._process_multipart_data(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 316, in _process_multipart_data
    self._on_request_body_finish()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 468, in _on_request_body_finish
    self._new_body += octoprint.util.to_bytes(p) + b'\r\n'
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
2019-11-25 17:11:29,612 - tornado.application - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 584, in _read_fixed_body
    ret = delegate.data_received(body)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/httpserver.py", line 311, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/routing.py", line 248, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2087, in data_received
    return self.handler.data_received(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 271, in data_received
    self._process_multipart_data(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 316, in _process_multipart_data
    self._on_request_body_finish()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 468, in _on_request_body_finish
    self._new_body += octoprint.util.to_bytes(p) + b'\r\n'
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
```

Lantoit, 2019-11-25:
Basic functionality tested with Pi3+ and sharebot NG (rumba+merlin)

"All is working fine" with Python 2.7. We need to test it also with Py 3.x??

Bye!

sebaminguez, 2019-11-26:
> Installed on my Ubuntu 18 Server System with Klipper has FW. >> Installation OK and System is running. Later i perform some prints.
> 
> Edit: octoprint_fanspeedslider doesnt work anymore.
> 
> ```
> 2019-11-25 17:59:12,930 - octoprint.plugin - ERROR - Error while calling plugin fanspeedslider
> Traceback (most recent call last):
>   File "/home/nas/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 224, in call_plugin
>     result = getattr(plugin, method)(*args, **kwargs)
>   File "/home/nas/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_fanspeedslider/__init__.py", line 19, in on_after_startup
>     self.get_settings_updates()
>   File "/home/nas/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_fanspeedslider/__init__.py", line 90, in get_settings_updates
>     self.defaultFanSpeed = self._settings.getInt(["defaultFanSpeed"])
>   File "/home/nas/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 521, in __getattr__
>     orig_func = decorator(orig_func)
>   File "/home/nas/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 134, in decorator
>     func.__qualname__ = to_native_str('warning_decorator_factory')
> AttributeError: 'instancemethod' object has no attribute '__qualname__'
> ```

The same for filamentmanager
`2019-11-25 22:54:33,413 - octoprint.plugin.core - ERROR - Exception while initializing plugin filamentmanager, disabling it
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1439, in initialize_implementation
    implementation.initialize()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_filamentmanager/__init__.py", line 53, in initialize
    self.filamentOdometer.set_g90_extruder(self._settings.getBoolean(["feature", "g90InfluencesExtruder"]))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 521, in __getattr__
    orig_func = decorator(orig_func)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 134, in decorator
    func.__qualname__ = to_native_str('warning_decorator_factory')
AttributeError: 'instancemethod' object has no attribute '__qualname__'`

gferon, 2019-11-26:
> hello @foosel ,
> install ok but pb with cura octoprint plugin, when i try to upload a gcode file from cura :
>
> ```
>
> 2019-11-25 17:09:34,089 - tornado.application - ERROR - Uncaught exception
> Traceback [...]

Same problem here, file upload with the API seems broken with Python 3.

foosel, 2019-11-26:
@gferon @devildant please open a full bug report, I cannot reproduce this right now and need more data. Uploading through the UI (exact same API endpoint) works.

@sebaminguez @schnello I can reproduce *that* and it's caused by a Python3 compat change plus those plugins still using deprecated methods. Looking into it.

foosel, 2019-11-26:
> @gferon @devildant please open a full bug report, I cannot reproduce this right now and need more data. Uploading through the UI (exact same API endpoint) works.

Update to that, I've since been able to reproduce it and it seems that missing optional headers in the upload request are the reason here. I've pushed a fix.

> @sebaminguez @schnello I can reproduce _that_ and it's caused by a Python3 compat change plus those plugins still using deprecated methods. Looking into it.

I've pushed a fix for that too.

foosel, 2019-11-26:
Now it's fixed.

sebaminguez, 2019-11-26:
@foosel awesome! I deep diving in the code but my python skills were not enough :D
Let me know If I can help with testing or anything else.

schnello, 2019-11-27:
Looks like the autologin function is not always working.
`
accessControl:
    autologinAs: admin
    autologinLocal: true
    localNetworks:
    - 127.0.0.1/8
    - 192.168.1.0/24
`

[image]

Pre 1.4 i need never any login informations with the settings above. Now with 1.4 i have to enter the login information once per day.

JohnOCFII, 2019-11-27:
Potential time-lapse issue with new libx264 - see #3353

foosel, 2019-11-28:
@schnello Created a ticket for that in #3357 

@JohnOCFII analysed and fixed, thanks!

FormerLurker, 2019-12-01:
I'm having a problem loading javascript in the 1.4.0rc1 and the latest staging/devel branch as well.

here is my console output:

```
(index):1 Refused to execute script from 'http://localhost:5000/static/webassets/packed_libs.js?3339d949' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.
(index):1 Refused to execute script from 'http://localhost:5000/static/webassets/packed_client.js?6627b31c' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.
(index):1 Refused to execute script from 'http://localhost:5000/static/webassets/packed_plugins.js?69c93951' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.
(index):1 Refused to execute script from 'http://localhost:5000/static/webassets/packed_core.js?eb76ab5f' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.

```

I've tried with both python 2.7 and 3.7, and in the latest release versions of Chrome, Firefox and Firefox Development Edition.  Let me know if you need any additional info.

FormerLurker, 2019-12-01:
Ok, I figured it out, but I have no idea how this was caused.  According to [this](https://github.com/pallets/flask/issues/1045) issue, the mime types per extension on a windows machine are stored within the registry.  Somehow this got changed to text/plain.  Since it was working fine before upgrading to Python 3.7 and 1.4.0rc1, I believe it must have had something to do with the switch to python 3.7.  I don't see how OctoPrint could be responsible for this in any way.  Please disregard my previous comment since it is windows specific and seemingly unrelated to V1.4.0rc1.

foosel, 2019-12-02:
1.4.0rc2 is out.

closed 2019-12-02 by foosel