OctoPrint/OctoPrint #3349: [1.4.0rc1] Unset optional header fields in upload parts can cause exceptions

foosel, 2019-11-26:
### Problem

If a part's filename, content type or such are unset on the upload, the corresponding field will stay at `None`. Trying to put that into the rewritten request body will cause an exception due to incompatible operands:

```
2019-11-25 17:09:34,089 - tornado.application - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 584, in _read_fixed_body
    ret = delegate.data_received(body)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/httpserver.py", line 311, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/routing.py", line 248, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2087, in data_received
    return self.handler.data_received(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 271, in data_received
    self._process_multipart_data(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 316, in _process_multipart_data
    self._on_request_body_finish()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 468, in _on_request_body_finish
    self._new_body += octoprint.util.to_bytes(p) + b'\r\n'
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
```

See also https://github.com/foosel/OctoPrint/issues/3347#issuecomment-558254959 and https://github.com/foosel/OctoPrint/issues/3347#issuecomment-558561001

### Solution

Do not try to write `None` values into the new body in the first place

gferon, 2019-11-26:
I had a smell that this would be about the HTTP headers. Thanks for fixing it so quickly!

devildant, 2019-11-26:
thx @foosel 
sorry i did not have time to answer before (i'm at work :))

foosel, 2019-12-02:
1.4.0rc2 is out.

closed 2019-12-02 by foosel

stevvie, 2019-12-10:
RC2 fixed the 504 error for cura but still getting 'command error:api/login?passive=true(400)' every few seconds in printoid with the new 14RC1 or 14RC2

foosel, 2019-12-10:
@stevvie can you please open a new ticket for that, that's a completely different error and from the looks of it more a case of printoid sending a malformed request that I'd like to get to the bottom of.

foosel, 2019-12-10:
@stevvie never mind, done

commit 471dab94 (2019-11-26): 🐛 Don't try to write None values into upload body

Closes #3349