OctoPrint/OctoPrint #3351: Multiple browser better updates [request]

elvinhaak, 2019-11-27:
I have generally multiple browsers open (on multiple computers) for the same printers and have somehow 'old info' on the browsers.

I have for example a PC in the livingroom and also one  near the printer and if I go to bed one browser running on my mobile.
When I have for example paused from the PC in the shed near the printer, the other browsers keep on writing ' printing' instead of changing to 'pausing' or 'paused'.
Also the case if I cancel a print: the other browsers keep on saying 'printing'.
Even if I start a new print, the old filename is still on top.

If I press 'refresh' on the other browsers, it is generally the same status, no changes. 
For example it says (internalnetworkaddressIP)#temp and it is still printing the file, even after the refresh. Also after a [CTRL]-[F5]. And I have done a cancel from one of the other connected computers on the network and for example used the control-tab to change the position to change the filament.
If I go back to the main page by typing the internalnetworkaddressIP in the browsers address, then I have the new info after a while waiting.
It can be also real old data from a couple of days ago if there is no active refresh (and no, no screensafer, the browser stays active on the PC or laptop, not used for anything else)

This happens to me on multiple Octoprint instances (multiple printers, each their own Raspberry-Pi), both with a PiZeroW as with a Pi3B.
Using multiple browsers from multiple PC's also seems to slow down the refresh of the pages and the reload is taking quite long, htop (with putty) is showing a high processor load in these cases. It seems that the Pi is using a lot of processor-power to write the same info for each browser (computer) that is looking at the same info.

Maybe it is possible to have the pages in a cache on the pi that is updated internally and only that cache-page is loading from the browsers instead of generating all data again per browser?

It is nice that it is possible to break off a print because if is clear something is wrong from the livingroom (looking at a webcam, not using the screen in octoprint, that is not possible with all cameras) and do a continuation from the shed's PC / laptop but it is at least confusing and taking a lot of time when that is still saying 'printing'. 
Somehow it is also very slow to refresh this way (probably of the high processor-usage), can take up to minutes to refresh (on the Pi3b, even longer on the PiZeroW).

Actually, part of the page seems to be refreshing (like the temperatures-info) but the rest of the page including the buttons are not in these cases.
The browser that has done the last changes (like pausing or the restart) has the right info and buttons, the others (can be 3 other computers) have not-updated info even after the [F5].

Screenshot is from one PC, example of print with the plugins that are loaded (on the Pi3b). There is no connection to access the pi from outside the internal-network (other then realvnc if vnc is started). Browsers used are mostly firefox but also chrome and epiphany. Makes no difference. PC'\s are multiple Linux-systems, mobile is Android 7.
[image]

foosel, 2019-11-28:
The behaviour you are describing is absolutely not normal and sounds more like a network issue or a bug than anything else. Normally OctoPrint's interface should always reflect the internal state accurately, without any hard reloads being necessary. If it doesn't, something is getting lost or cached somewhere it shouldn't.

I suggest you get in touch on this on the [community forum](https://community.octoprint.org) so we can rule out network issues first.

closed 2019-11-28 by foosel

elvinhaak, 2019-11-28:
Hi. Ok thanks for the answer. I will try to open it on the forum.

Elvin