OctoPrint/OctoPrint #3354: [1.4.0rc1] Time-lapse fails with libx264 selected - "WARNING - Cannot capture image, image number is unset"

JohnOCFII, 2019-11-27:
#### What were you doing?

Printing from OctoPrint to Prusa MK3S with MMU2S.  I had configured Time-lapse to use the new libx264 in `Settings.`. No time-lapse was created.  No indication of time-lapse activity.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

Expected time-lapse to be created

#### What happened instead?

No time-lapse was created.

#### Did the same happen when running OctoPrint in safe mode?

Did not test in safe mode, but will switch back to mpeg2 and test again next.

#### Version of OctoPrint

1.4.0rc1

#### Operating System running OctoPrint

OctoPi 0.16.0

#### Printer model & used firmware incl. version

Prusa MK3S with MMU2S with firmware 3.8.1

#### Browser and version of browser, operating system running browser

macOS Catalina (10.15.1) with Safari 13.0.3

#### Link to octoprint.log

https://gist.github.com/JohnOCFII/750a3d0d0afc18bfaed0e3fee2c589ef

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

JohnOCFII, 2019-11-28:
Same results, same message in OctoPrint.log after switching back to mpeg2.  Will try Safe Mode next.

JohnOCFII, 2019-11-28:
Failing in safe mode too.  This looks suspect:

```2019-11-27 19:01:28,486 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: Printer_Tools/Support_Test_Stand_t3_0.2mm_PLA_MK3SMMU2S.gcode, owner: osh, user: osh
2019-11-27 19:01:28,489 - octoprint.events - ERROR - Got an exception while sending event PrintStarted (Payload: {'origin': u'local', 'name': u'Support_Test_Stand_t3_0.2mm_PLA_MK3SMMU2S.gcode', u'user': 'osh', u'owner': 'osh', 'path': u'Printer_Tools/Support_Test_Stand_t3_0.2mm_PLA_MK3SMMU2S.gcode', 'size': 373877L}) to <bound method ZTimelapse.on_print_started of <octoprint.timelapse.ZTimelapse object at 0x72d5b330>>
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/events.py", line 187, in _work
    listener(event, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/timelapse.py", line 468, in on_print_started
    self.start_timelapse(payload["file"])
KeyError: u'file'```

Full OctoPrint.log here: https://gist.github.com/JohnOCFII/e080c299727c8524bcd7bd81cf025649

loskexos, 2019-11-28:
I've got the exact same issue:
- timelapses no longer created since 1.4.0rc1
- octoprint.log shows:

```
2019-11-28 09:44:09,518 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2019-11-28 09:44:09,551 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: mk3_mk3s_upgrade_allesdruckenufeinmal_PETG_0.05mm_PET_MK3_19h54m.gcode, owner: octoadmin, user: octoadmin
2019-11-28 09:44:09,553 - octoprint.events - ERROR - Got an exception while sending event PrintStarted (Payload: {'origin': u'local', 'name': u'mk3_mk3s_upgrade_allesdruckenufeinmal_PETG_0.05mm_PET_MK3_19h54m.gcode', u'user': 'octoadmin', u'owner': 'octoadmin', 'path': u'mk3_mk3s_upgrade_allesdruckenufeinmal_PETG_0.05mm_PET_MK3_19h54m.gcode', 'size': 35547602L}) to <bound method TimedTimelapse.on_print_started of <octoprint.timelapse.TimedTimelapse object at 0x704fb510>>
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/events.py", line 187, in _work
    listener(event, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/timelapse.py", line 782, in on_print_started
    Timelapse.on_print_started(self, event, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/timelapse.py", line 468, in on_print_started
    self.start_timelapse(payload["file"])
KeyError: u'file'
2019-11-28 09:44:09,628 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2019-11-28 09:44:09,690 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
```

Already tried different encoding format, disable plugins, test the settings. I've got two webcams configured if that is relevant.
Error stack looks like more syntax-ish though (looking around in the code without python knowledge: key file ... if that should reference the gcode file name then the correct key might be gcodeFile). Need to wait until tomorrow to test (ongoing print).

foosel, 2019-11-28:
Ah shoot. That looks like an issue due to deprecating something and for some reason *not* catching all places where it was used.

foosel, 2019-11-28:
Fix should be ready for 1.4.0rc2

foosel, 2019-12-02:
1.4.0rc2 is out.

commit 59e39662 (2019-11-28): 🐛 timelapse: fix accessed event payload field

Closes #3354
closed 2019-12-02 by foosel