OctoPrint/OctoPrint #3357: [1.4.0] Local autologin not working

foosel, 2019-11-28:
### Problem

As reported by @schnello in https://github.com/foosel/OctoPrint/issues/3347#issuecomment-559239539:

> Looks like the autologin function is not always working.
> `accessControl: autologinAs: admin autologinLocal: true localNetworks: - 127.0.0.1/8 - 192.168.1.0/24`
> 
> [image]
> 
> Pre 1.4 i need never any login informations with the settings above. Now with 1.4 i have to enter the login information once per day.

### Solution

Needs further analysis first.

loskexos, 2019-11-28:
Not sure if I observe the same item with this:

- logon (as user octoadm)
- Then close browser (alt-F4)
- Then open browser again and open the octoprint webpage

result: I have to logon by manually entering user and password.

I thought it's a security feature :). But looks similar to what is described here.

The log states it actually is logging on a user passively. Username "None".

<<action: open webpage>>
```
2019-11-28 21:35:20,132 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.33.129
2019-11-28 21:35:20,393 - octoprint.server.util.flask - INFO - Passively logging in user None from ::ffff:192.168.33.129
```
<<action: enter username octoadmin/password>>
```
2019-11-28 21:35:29,765 - octoprint.server.api - INFO - Actively logging in user octoadm from ::ffff:192.168.33.129
2019-11-28 21:35:31,567 - octoprint.server.util.sockjs - INFO - User octoadm logged in on the socket from client ::ffff:192.168.33.129
```
code for the message at "21:35:20,393" seems to be in ./oprint/lib/python2.7/site-packages/octoprint/server/util/flask.py, lines 593-595. (only matching location for that message above I found).
```
        if user is not None and user.is_active:
                # login known user
                logger.info("Passively logging in user {} from {}".format(user.get_id(), remote_address))
                user = login(user)

```
Interestingly code checks IF user is **NOT** "None" and then  logs the user as "None" in octoprint.log. Are "if user" and "user.get_id()" intentionally two different things? 
(I don;t know python, thats how understand that line. Maybe it should read ..if user.get_id() is not None... ?).

It's not even reaching the autologin parts for me  in the following lines that @schnello reported, so maybe a different case.

Just an observation reading this thread. Not a problem for me.
Observed with chrome and firefox.

foosel, 2019-11-29:
@loskexos that's a good hint actually... there might be a logic issue here in that code block due to anonymous users no longer being `None` but having no id. Hence the passive login will trigger even for them and the autologin won't even initiate. 

I'll have to take a close look at things here, I'm fairly certain there was a reason why I don't check if the user is anonymous here, but I'm not 100% sure.

foosel, 2019-11-29:
Should be fixed by the above commit.

schnello, 2019-11-29:
tested and i can confirm. Autologin is now working again.

foosel, 2019-12-02:
1.4.0rc2 is out.

commit 00d03964 (2019-11-29): 🐛 Fix autologin

Closes #3357
closed 2019-12-02 by foosel