OctoPrint/OctoPrint #3358: [1.4.0rc1] Error when uploading through Cura plugin

fieldOfView, 2019-11-29:
#### What were you doing?
Slicing and uploading a gcode file from Cura

#### What did you expect to happen?
Upload without a hitch

#### What happened instead?
Upload seems to happen, but OctoPrint returns a 502 Bad gateway error and the file does not appear in the files list.
In the logs, I see the following exception:
```
2019-11-28 17:53:45,307 - tornado.application - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/http1connection.py", line 584, in _read_fixed_body
    ret = delegate.data_received(body)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/httpserver.py", line 311, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/routing.py", line 248, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado/web.py", line 2087, in data_received
    return self.handler.data_received(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 271, in data_received
    self._process_multipart_data(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 316, in _process_multipart_data
    self._on_request_body_finish()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 468, in _on_request_body_finish
    self._new_body += octoprint.util.to_bytes(p) + b'\r\n'
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
```
Uploading the same file through the webbrowser works.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.4.0 rc 1

#### Operating System running OctoPrint
OctoPi 0.14 originally, which has since been upgraded to stretch

#### Printer model & used firmware incl. version
Not connected to printer for now

#### Link to octoprint.log
https://pastebin.com/gUbng6RW
(includes both normal and safe mode reproduction)

#### Link to contents of terminal tab or serial.log
na

#### Link to contents of Javascript console in the browser
na

#### Screenshot(s)/video(s) showing the problem:
na

I have read the FAQ.

foosel, 2019-11-29:
Duplicate of #3349 ;)

closed 2019-11-29 by foosel

fieldOfView, 2019-11-29:
Sorry I missed that.

foosel, 2019-11-29:
No worries, happens. Thanks for reporting!