OctoPrint/OctoPrint #3365: [RC 1.4.0rc2]global API KEY don't have all permission

devildant, 2019-12-02:
#### What were you doing?
following the RC update, I see that it is no longer possible to call the API of plugins.
i take an error 403.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

exemple with shutdown printer plugin : 
url : /api/plugin/shutdownprinter
body : {"command":"enable", "eventView" : true}
API key in header : X-Api-Key=666666......

result : 
http code:  403
body : Insufficient rights

inside method on_api_command (shutdown printer plugin) i put this control : 
```
def on_api_command(self, command, data):
                if not user_permission.can():
                        return make_response("Insufficient rights", 403)
```

but in 1.3.12 no pb

I also try with user API it's work correctly, but with global api key no 
#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

devildant, 2019-12-02:
I try to replace the permission control and it seems to work
```
if not user_permission.can():
                        return make_response("Insufficient rights", 403)
```
become
```
 user_id = current_user.get_name()
                if not user_id or not admin_permission.can():
                        return make_response("Insufficient rights", 403)
```

but inside the octoprint code i see that : 
```
# we set admin_permission to a GroupPermission with the default admin group
admin_permission = octoprint.util.variable_deprecated("admin_permission has been deprecated, "
                                                      "please use individual Permissions instead",
                                                      since="1.4.0")(groups.GroupPermission(groups.ADMIN_GROUP))

# we set user_permission to a GroupPermission with the default user group
user_permission = octoprint.util.variable_deprecated("user_permission has been deprecated, "
                                                     "please use individual Permissions instead",
                                                     since="1.4.0")(groups.GroupPermission(groups.USER_GROUP))
```

what is the right solution for controlling permissions on plug-in APIs?

foosel, 2019-12-03:
> what is the right solution for controlling permissions on plug-in APIs?

Declaring them for your plugins and using them instead of the generic and deprecated admin/user permissions. There's a new hook for that called `octoprint.access.permissions` (which I just realized I forgot to document, so thanks for that reminder!) and you can see how that is used in the various bundled plugins, e.g. [the appkeys plugin](https://github.com/foosel/OctoPrint/blob/rc/devel/src/octoprint/plugins/appkeys/__init__.py).

Still, the global API key *should* indeed always work, so I'll look into that as well here.

foosel, 2019-12-03:
Should be fixed and I also added the missing docs.

devildant, 2019-12-03:
thx @foosel , juste a little question, 
I just have a comment, it missing in the documentation the import necessary for use permission hook, and there will be problems of backward compatibility with version 1.3.12 (for those who will not update octoprint)
I fix this in the example below:
```
# coding=utf-8
from __future__ import absolute_import
from flask import make_response

from octoprint.server import NO_CONTENT, current_user, admin_permission
#for retrocompatibility
try:
	from octoprint.access.permissions import Permissions, ADMIN_GROUP, USER_GROUP
except (ImportError, RuntimeError):
	from octoprint.server import user_permission

from flask_babel import gettext

import octoprint.plugin

class TestPlugin(octoprint.plugin.StartupPlugin,
                       octoprint.plugin.TemplatePlugin,
                       octoprint.plugin.SettingsPlugin,
					   octoprint.plugin.SimpleApiPlugin,
                       octoprint.plugin.AssetPlugin):
	def on_after_startup(self):
		self._logger.info("test init")

	def get_api_commands(self):
		return dict(isco=[])
	
	def get_additional_permissions(self):
		return [
				dict(key="ADMIN",
					name="test plugin",
					description=gettext("Allows to access to api of plugin test."),
					default_groups=[ADMIN_GROUP],
					roles=["admin"],
					dangerous=True)
				]	
				
	def on_api_command(self, command, data):
		#for retrocompatibility
		try:
			plugin_permission = Permissions.PLUGIN_TEST_ADMIN.can()
		except:
			plugin_permission = user_permission.can()
		if not plugin_permission:
			return make_response("Insufficient rights", 403)
			
	def get_settings_defaults(self):
		return dict()

	def get_template_configs(self):
		return [
			dict(type="settings", custom_bindings=False)
		]

	def get_assets(self):
		return dict(
			js=["js/test.js"],
			css=["css/test.css"]
		)

__plugin_name__ = "Test"

def __plugin_load__():
	global __plugin_implementation__
	__plugin_implementation__ = TestPlugin()

	global __plugin_hooks__
	__plugin_hooks__ = {
		"octoprint.access.permissions": __plugin_implementation__.get_additional_permissions
	}

```

devildant, 2019-12-06:
it's just a small recommendation for the doc :)

foosel, 2019-12-09:
I know. I'd rather have plugin authors encourage people to update to newer OctoPrint versions though ;) I'll think about it, currently still hunting down a few reported bugs, that takes priority.

foosel, 2019-12-12:
1.4.0rc3 has just been released

commit 2200f218 (2019-12-03): 🐛 Fix permissions of global API key

Closes #3365
closed 2019-12-12 by foosel