OctoPrint/OctoPrint #3366: [1.4.0rc2] Appkeys request returns 500/Internal server error

fieldOfView, 2019-12-02:
#### What were you doing?
I am trying to request an appkey.

```
curl -X POST -H "Content-Type: application/json" http://octopi.local:80/plugin/appkeys/request -d "{\"app\": \"Cura\"}"
```

#### What did you expect to happen?
I expect a 201 response and a Location header so I can continue with the appkey request and probing

#### What happened instead?
The curl command returns the following:
```
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">                                                                                                 
<title>500 Internal Server Error</title>                                                                                                                
<h1>Internal Server Error</h1>                                                                                                                          
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>                                                                                                                                                 
```
The octoprint.log only shows the following error log, not why the error happened
```
2019-12-02 22:10:09,980 - tornado.access - ERROR - 500 POST /plugin/appkeys/request (25.39.73.122) 132.71ms
```

#### Did the same happen when running OctoPrint in safe mode?
yes

#### Version of OctoPrint
1.4.0 rc2

#### Operating System running OctoPrint
OctoPi 0.14, which has been updated to Stretch

#### Printer model & used firmware incl. version
na

#### Browser and version of browser, operating system running browser
na

#### Link to octoprint.log
https://pastebin.com/Gwk0QWqP

#### Link to contents of terminal tab or serial.log
na

#### Link to contents of Javascript console in the browser
na

#### Screenshot(s)/video(s) showing the problem:
na

foosel, 2019-12-03:
Great. My development environment was still on Py3 for an analysis, and this particular issue seems to be Py2 only, hence it slipped through. That's going to be fun for the next year or so...

foosel, 2019-12-12:
1.4.0rc3 has just been released

commit b57b7cb6 (2019-12-03): 🐛 appkeys: fix key gen under Python 2

And refactor key generation to be located in ONE central utility method
to avoid issues like that in the future.

Closes #3366
closed 2019-12-12 by foosel