OctoPrint/OctoPrint #3367: Windows 10; 1.14.0RC2 fails to load UI

elfelton, 2019-12-02:
#### What were you doing?

navigating to localhost:5000 shows " Loading OctoPrint's UI, please wait..." forever.

From Chrome's console log I think this is the root-cause:

Refused to execute script from 'http://localhost:5000/static/webassets/packed_libs.js?3339d949' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.
(index):1 Refused to execute script from 'http://localhost:5000/static/webassets/packed_client.js?92e73cbf' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.
(index):1 Refused to execute script from 'http://localhost:5000/static/webassets/packed_core.js?a90e3d70' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.
(index):6482 Uncaught ReferenceError: $ is not defined
at (index):6482

Background...

I installed Python 3.8 for Windows 10
I installed OctoPrint 1.4.0rc1 to c:\octoprint using the setting-up-octoprint-on-windows guide.

I upgraded to 1.4.0rc2 on Dec 2

I tried using chrome and edge browsers.

#### What did you expect to happen?
I expected to see the standard OctoPrint UI (or on the very first run, see the setup wizard)

#### What happened instead?
I see "Loading OctoPrint's UI, please wait..."
And no other indication of anything happening

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint
1.4.0rc1 and 1.4.0rc2

#### Operating System running OctoPrint
Windows 10 Pro 1809

2019-12-02 13:55:02,752 - octoprint.environment - INFO - Detected environment is Python 3.8.0 under Windows (win32). Details:
|  hardware:
|      cores: 8
|      freq: 2401.0
|      ram: 17057427456
|  os:
|      id: windows
|      platform: win32
|  python:
|      pip: 19.3.1
|      version: 3.8.0
|      virtualenv: c:\octoprint\venv
#### Printer model & used firmware incl. version

Currently, the Windows machine is not physically connected to a printer.

#### Browser and version of browser, operating system running browser

Chrome Version 78.0.3904.108 (Official Build) (64-bit)
Microsoft Edge 44.17763.831.0

#### Link to octoprint.log
https://gist.github.com/elfelton/7c24f0d6b1834c538206fb997e759309#file-octoprint-log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

Refused to execute script from 'http://localhost:5000/static/webassets/packed_libs.js?3339d949' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.
(index):1 Refused to execute script from 'http://localhost:5000/static/webassets/packed_client.js?92e73cbf' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.
(index):1 Refused to execute script from 'http://localhost:5000/static/webassets/packed_core.js?a90e3d70' because its MIME type ('text/plain') is not executable, and strict MIME type checking is enabled.
(index):6530 Uncaught ReferenceError: $ is not defined
    at (index):6530

#### Screenshot(s)/video(s) showing the problem:

https://user-images.githubusercontent.com/8985461/70002202-35f7ed00-1557-11ea-8885-56f2e896d17e.png

elfelton, 2019-12-02:
Hello, I am able to get OctoPrint 1.3.12 and Python 2.7 running on the same machine.  
The octoprint.log will show both 1.4.0 and 1.3.12 information over the last several days.  The 1.3.12 UI starts up as expected, I am only encountering this issue with 1.4.0

elfelton, 2019-12-03:
I'm not sure what information I'm missing that the GitIssueBot is flagging.  I provided everything I can given the current situation.  If a human has any specific questions I'm happy to answer them.

elfelton, 2019-12-03:
I found this issue and resolution elsewhere
https://github.com/jupyterlab/jupyterlab/issues/6098
I updated my Registry and the UI now loads in Chrome.

I don't know if the proper thing to do is close this OctoPrint issue or to document the behavior and the solution.  Or have the OctoPrint installer check the registry and make the appropriate changes.

foosel, 2019-12-03:
I've added a bit of protection against this causing issues by enforcing the correct MIME types for .js and .css files on the built-in webserver, regardless of whether the registry has broken values configured or not.

Still interesting that this seems to be widespread enough that there's a whole number of tickets on various projects out there. I wonder what kind of software is behind these messed up registries. My own was clean, I borked it intentionally to test the above commit's effect how ever and can confirm it worked.

foosel, 2019-12-12:
1.4.0rc3 has just been released

commit 0c770149 (2019-12-03): ☔ Make sure js/css mime types are always correct

Even if we are running on windows and something is borked in the
registry...

Closes #3367
closed 2019-12-12 by foosel