OctoPrint/OctoPrint #3368: GCODE viewer shows the wrong layer

RomRider, 2019-12-03:
#### What were you doing?

1. Go to GCODE viewer
2. Wait for it to load, it displays the wrong layer.

It's usually ahead of time by a lot of layers (depending on the model size)

#### What did you expect to happen?
It should be in sync with the current layer being printed.

#### What happened instead?
It shows a layer which will be printed in the future 

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.13.12

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
Klipper (same with Marlin) / Ender 5

#### Browser and version of browser, operating system running browser
Chrome (latest), MacOS

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2019-12-03:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * your [exact version of OctoPi](https://discourse.octoprint.org/t/how-can-i-find-out-the-version-of-octoprint-or-octopi-i-am-running/204/1)
  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * contents of the terminal tab, or even better a [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * since this looks like an issue related to specific GCODE file properties, one or more GCODE files with which to reproduce this

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

RomRider, 2019-12-03:
I'm using OctoPi `0.17.0`

There's nothing relevant in serial.log but I think that the gcode is the reason it behaves like this:
[CE5_Stand.gcode.txt](https://github.com/foosel/OctoPrint/files/3917157/CE5_Stand.gcode.txt)

* I draw a line to clean the nozzle at the beginning, this line is 0.28 in height (considered as layer 2 in the gcode viewer, although it is layer 1 and the very first thing that happens)
* Then the skirt starts to print and is 0.20mm in height (should still be layer 1)

Maybe that's what confuses the gcode viewer?

Also attached the octoprint.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/3917221/octoprint.log)

foosel, 2019-12-03:
So, there's two things to consider here. 

One is that the GCODE viewer will always be a bit ahead of what is currently being printed due to buffering that happens on firmware side. If you are just before a layer change the next layer might be displayed because the corresponding commands have already been sent to the printer and there's no way of knowing when they will actually be executed.

The other thing is that the GCODE viewer by default sorts the layers by height, ignoring empty layers. So the numbers won't necessarily match the output from the third party plugin you are referring to in your screenshot there. You can change the viewer's behaviour in the advanced settings though:

[image]

I suggest to first experiment with that and see if that maybe already fixes the observed inconsistency for you. If not I'll have to dig deeper.

RomRider, 2019-12-03:
Thanks @foosel. It is at least 7 to 8 layers ahead in every print (I use Cura as a slicer). I already played with those advanced options and it is always far ahead by many layers compared to the actual print.

It looks like issue #3272 but with way more difference in layers. I don't know what is the buffer size of klipper but I'm pretty sure it's less than 1 full layer for the above gcode.

TheDevilOnLine, 2020-03-13:
Min is actually behind most of the time (sometimes even by 15 layers on a 220 layer full plate print on an Ender 3 Pro) .

foosel, 2020-04-14:
Just for the record, this ticket is about the viewer being *ahead* of the actual printing progress (which to a certain degree is normal due to firmware side buffers that OctoPrint cannot monitor). It is *not* about the viewer *lagging severely behind* the actual progress. 

**If you found this ticket due to your gcode viewer lagging behind**, specifically starting with OctoPrint 1.4.0, then please take a look at [this post on the forum](https://community.octoprint.org/t/curious-issue-with-print-progress/16304/60?u=foosel) and install the plugin linked therein.

foosel, 2020-06-12:
I have to mark this as unreproduced for now - I cannot reproduce it, and I don't see anything wrong with the position calculation. Maybe it is a huge buffer on the printer side, it's impossible for me to determine right now. Will have to see if we get more reports.

ntrp, 2020-08-07:
I am seeing the same behavior after upgrade to 1.4.0, before it was working fine. The Gcode viewer starts at 4 by default and is always 4 layer ahead... I think it was happening with Prusa Slic3r and now I switched to Cura and it still happens. If needed I might collect more info, if there anything special to check to debug the issue let me know.

foosel, 2020-08-07:
Quoting the [Contribution Guidelines](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report):

> What should I include in a bug report?
> 
> First of all make sure your use a descriptive title. "It doesn't work" and similar unspecific complaints are NOT descriptive titles.
> 
> **Always use the following template, even if only adding a "me too" to an existing ticket:**

If I haven't been able to fix or even reproduce it up until now, that won't change by someone chiming in saying "oh, I have this issue too". What I need is more information to be able to narrow in on it.

ntrp, 2020-08-07:
I really don't get your point... That's why I asked what could I analyze to make it easier or to provide more info.. Also I said that it happened after the upgrade and that it starts ahead from the beginning so I didn't just write "me too"

foosel, 2020-08-07:
My point is that I need all the info from you I can get. That means, logs, whether the issue persists in safe mode, exact version of OctoPrint, OctoPi, browser and such, GCODE files against which to reproduce etc. Just saying you are experiencing the issue as well doesn't help. You wrote "if need me I might collect more info" and I quoted the part from the contribution guidelines that tells you that yes, I in fact do need more info because I haven't been able to figure this out with the info I so far have.

In short, when you say you have an issue as well, share everything in the ticket template as well. I'd rather have too much info than too little. I get that some of you are seeing an issue, but I can't see it and that means I require as much information as I can get from anyone experiencing this issue to be able to figure it out and ideally fix it.

closed 2021-03-31 by LazeMSS