OctoPrint/OctoPrint #3369: Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://192.168.178.30:8080/sockjs-node/info?t=1575365178459. (Reason: CORS request did not succeed)

Wulfheart, 2019-12-04:
Hello everybody,

I'd like to connect my custom Vuejs client with a websocket. Unfortunately this doesn't work. I have enabled CORS support in the API section.
It doesn't work with socket.io-client nor with sockjs even though it works with some websocket clients like `Simple WebSocket Client`.

Here are some error messages:

```
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://localhost:8080/sockjs-node/info?t=1575366607181. (Reason: CORS request did not succeed).
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://192.168.178.30:8080/sockjs-node/info?t=1575366609441. (Reason: CORS request did not succeed).
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://localhost:8080/sockjs-node/info?t=1575366612157. (Reason: CORS request did not succeed).
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://localhost:8080/sockjs-node/info?t=1575366793056. (Reason: CORS request did not succeed).
Cross-Origin Request Blocked: The Same Origin Policy disallows reading the remote resource at http://localhost:8080/sockjs-node/info?t=1575367310669. (Reason: CORS request did not succeed).
```

How can I fix this aside from doing periodical request with the API?

Thank you in advance.

[Forum Post](https://community.octoprint.org/t/cross-origin-request-blocked-the-same-origin-policy-disallows-reading-the-remote-resource-at-http-192-168-178-30-8080-sockjs-node-info-t-1575365178459-reason-cors-request-did-not-succeed/13990/2)

foosel, 2019-12-04:
Hey @Wulfheart, when I told you to file a ticket I also meant to fill out the full ticket template, because that way I get my hands on some crucial information I need in order to look into this which right now I still don't have. So please do that :) Additionally, as requested in the forum post, please provide some client code with which to reproduce this.

Wulfheart, 2019-12-04:
I'm sorry, I will provide more info later even though I don't have the socket code anymore.

foosel, 2019-12-06:
I really need some code to reproduce this with, otherwise it'll get a bit hard to analyse what *exact* issue you ran into, or make sure that my reproduction steps are correct.

closed 2019-12-19 by GitIssueBot