OctoPrint/OctoPrint #3371: octoprint.util.comm - ERROR - Something crashed inside the serial connection loop

Lantoit, 2019-12-05:
#2651  What were you doing?

Trying to print from SD
I've selected a file and octoprint report 

"There was an unknown error while talking to your printer. Please consult the terminal output and octoprint.log for details. Error: See octoprint.log for details"

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.

#### What did you expect to happen?
Print from SD

#### What happened instead?
Error

#### Did the same happen when running OctoPrint in safe mode?
Same error
#### Version of OctoPrint

    OctoPrint 1.4.0rc2 running on OctoPi 0.17.0 

#### Operating System running OctoPrint
 on OctoPi 0.17.0 

#### Printer model & used firmware incl. version
Sharebot with Marlin 1.0.x upgraded yesterday

#### Browser and version of browser, operating system running browser
Ubuntu 19.10 + Firefox 

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

[serial.log.2019-12-05_18-34-19.log](https://github.com/foosel/OctoPrint/files/3928560/serial.log.2019-12-05_18-34-19.log)
[serial.log.2019-12-05_18-37-36.log](https://github.com/foosel/OctoPrint/files/3928561/serial.log.2019-12-05_18-37-36.log)

Lantoit, 2019-12-05:
Sorry...some log file appear missing. I retry.
[octoprint.log](https://github.com/foosel/OctoPrint/files/3928571/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/foosel/OctoPrint/files/3928572/plugin_pluginmanager_console.log)
[plugin_softwareupdate_console.log](https://github.com/foosel/OctoPrint/files/3928573/plugin_softwareupdate_console.log)
[serial.log](https://github.com/foosel/OctoPrint/files/3928574/serial.log)

foosel, 2019-12-06:
Thanks, that indeed looks like there's still a bug in there.

foosel, 2019-12-12:
1.4.0rc3 has just been released

commit 124f01ca (2019-12-06): 🐛 py3: ensure an int as rolling window size

Closes #3371
closed 2019-12-12 by foosel