OctoPrint/OctoPrint #3372: Getting 'pip command not found' error message when managing plugins

Erutan409, 2019-12-07:
#### What were you doing?

Navigated to `Plugin Manager` in configuration interface.

#### What did you expect to happen?

Be able to install/uninstall/activate/deactivate plugins.

#### What happened instead?

Received the following error:

```
The pip command could not be found or does not work correctly for this installation of OctoPrint - please consult the log file for details and if necessary configure it manually. No installation and uninstallation of plugin packages is possible while pip is unavailable.
```

#### Did the same happen when running OctoPrint in safe mode?

No.

#### Version of OctoPrint

1.3.12

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Chrome & Brave (latest versions)

#### Link to octoprint.log

[Log](https://gist.github.com/Erutan409/fdc9f52b00d0b736a6ab7ffc3cb7dcf8)

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

[JavaScript Console Output](https://gist.github.com/Erutan409/98e6824164e90d811772be31d835c503)

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

---

Interesting enough, though:

As I was going through the ticket creation process/guide, after I toggled between safe mode and back into normal mode, I can't reproduce the error, anymore.

This is weird, because I just reinstanced my raspberry pi (full format, etc. - using the official image) a couple weeks ago due to a similar issue I assumed was due to me using Octoprint since '16, doing regular automatic updates and breaking at some point.

I assume this should be looked into, but I'm not sure what the active devs on this project might have cooking in the background.

foosel, 2019-12-09:
It could be an issue either with loading the information from the API or processing it. Thing is, if something goes wrong retrieving the plugin information from the API endpoint, that message will be wrongfully displayed. In your log I don't see anything that looks like there has been an *actual* issue finding a usable `pip`, so my money is on that having happened here.

Sadly the only thing in your JS console log that I can see wrong is setting up a websocket connection which would not be able to cause this problem. So this is a bit of a tricky situation. Marking it as unreproduced for now, let's see if someone else might run into it and we can narrow it down maybe...

Was it reliably reproducible before switching to safe mode and back? Did it go away when switching to safe mode or only after switching back? If it went away immediately after switching to safe mode it could be some JS hiccup caused by some third party plugin, potentially cleared out with cache, but it's just blind guessing at this point.

Erutan409, 2019-12-09:
@foosel FWIW - in case it wasn't obvious: I removed my actual FQDN and replaced it with `redacted`, since mine is publicly accessible; albeit via a proxy that does some filtering and DDoS mitigation.

Erutan409, 2019-12-09:
@foosel It wasn't reproducible after toggling into safe mode and back.

Stady234, 2019-12-22:
A fresh install from the web download today (12/22/2019) on first boot of Octoprint, after setup... i get the same PIP message:

`The pip command could not be found or does not work correctly for this installation of OctoPrint - please consult the log file for details and if necessary configure it manually. No installation and uninstallation of plugin packages is possible while pip is unavailable.`

Quas7, 2020-02-23:
I faced the same issue on two machines.
I am not sure, what I did but on the newer installation from this year the error resolved itself.

On the still failing machine I updated the Spaghettidetective and had some strange behaviour afterwards and only then noticed, that pip was not working from within the web-interface. I uninstalled the it manually: https://community.octoprint.org/t/i-installed-a-broken-plugin-how-do-i-uninstall-it-again/217

`2020-02-22 11:02:36,366 - octoprint.plugins.thespaghettidetective - ERROR - Backing off 1.124928 seconds: Janus quit! This should not happen. Exit code: 127`

Failing to find pip on this machine:
|      model: Raspberry Pi 3 Model B Rev 1.2
|      octopi_version: 0.14.0
|    pip: 20.0.2
|    version: 2.7.9

and machine was fine with this setup the day before updating anything:
|      octopi_version: 0.14.0
|    pip: 10.0.1
|    version: 2.7.9

this newer machine is now running fine again (for whatever reason - not sure, if my "apt install python-pip" did anything here):
|      model: Raspberry Pi 3 Model B Rev 1.2
|      octopi_version: 0.14.0
|    pip: 19.3.1
|    version: 2.7.9

--> will now try to downgrade failing machine to 19.3.1, too.

a few logfile anomalies from the failing machine - maybe this helps someone identifying the root problem:
`
2020-02-23 23:14:16,436 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial port"
2020-02-23 23:14:16,439 - octoprint.plugin - ERROR - Error while calling plugin discovery
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 219, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery/__init__.py", line 143, in on_startup
    self.zeroconf_register("_http._tcp", self.get_instance_name(), txt_record=self._create_http_txt_record_dict())
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery/__init__.py", line 201, in zeroconf_register
    self._sd_refs[key] = pybonjour.DNSServiceRegister(**params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pybonjour.py", line 1132, in DNSServiceRegister
    None)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pybonjour.py", line 286, in _errcheck
    raise cls(result)
BonjourError: (-65537, 'unknown')
`

`2020-02-23 23:14:17,072 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2020-02-23 23:14:17,268 - octoprint.util.pip - ERROR - Error while discovering pip command
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/pip.py", line 154, in trigger_refresh
    self._setup_pip()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/pip.py", line 187, in _setup_pip
    pip_command, pip_sudo = self._get_pip_command()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/pip.py", line 441, in _get_pip_command
    return self.autodetect_pip(), False
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/pip.py", line 260, in autodetect_pip
    p = sarge.run(command + ["--version"], stdout=sarge.Capture(), stderr=sarge.Capture())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1462, in run
    p.run(input=input, async_=async_)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1071, in run
    self.run_node(node, input=input, async_=False)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge/__init__.py", line 662, in run
    self.stdin = p.stdin
UnboundLocalError: local variable 'p' referenced before assignment
2020-02-23 23:14:17,345 - octoprint.plugin - ERROR - Error while calling plugin softwareupdate
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 219, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 109, in on_after_startup
    self._check_environment()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 196, in _check_environment
    and get_comparable_version(versions["pip"]) >= get_comparable_version(MINIMUM_PIP)
[... cut ...]
TypeError: argument of type 'NoneType' is not iterable
2020-02-23 23:14:17,359 - octoprint.plugins.HeaterTimeout - INFO - Starting up...
`

and many of these in the end:
`2020-02-23 23:16:18,082 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2020-02-23 23:16:32,916 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
2020-02-23 23:16:33,952 - octoprint.plugins.forcelogin - WARNING - Socket message held back, but backlog full. Throwing message away: plugin
`

Quas7, 2020-02-23:
Just found the root-cause, I think.
pip, pip2 and pip2.7 in ~/oprint/bin currently are 0 bytes. 
Either my filesystem has broken down or the spaghetti-update went crazy and killed those files.

Stady234, 2020-02-24:
Has any postings on "spaghetti detective" github verify that the current version and use the new pip that does not support python2.7?
I ask due to not using that plugin.  It is past the Jan EOL Support for PIP vs Phython2.7 and below.

olivbd, 2020-07-06:
Same here on fresh install from image `2019-09-26-octopi-buster-lite-0.17.0.img`
[image]

The following files `pip`, `pip2` and `pip2.7` are in `~/oprint/bin` and are working in command line.

foosel, 2020-07-06:
This has since gotten an FAQ entry and is also resolved on current versions of OctoPrint.

https://community.octoprint.org/t/i-just-installed-octoprint-1-3-12-octopi-0-17-0-and-the-plugin-manager-cant-find-pip-whats-up/15692

closed 2020-07-06 by foosel

olivbd, 2020-07-06:
> This has since gotten an FAQ entry and is also resolved on current versions of OctoPrint.
> 
> https://community.octoprint.org/t/i-just-installed-octoprint-1-3-12-octopi-0-17-0-and-the-plugin-manager-cant-find-pip-whats-up/15692

Ok thanks! I confirm that is working now.