OctoPrint/OctoPrint #3374: 1.4.0rc3 exception -  Exception in callback functools.partial

gdombiak, 2019-12-07:
#### What were you doing?

Printing (from OctoPrint) using 1.4.0 rc3 (latest from github staging/devel branch)

#### What did you expect to happen?

No exception in the logs

#### What happened instead?

Found the exception below while tailing octoprint.log.  

> 2019-12-07 22:25:16,718 - tornado.application - ERROR - Exception in callback functools.partial(<function wrap.<locals>.null_wrapper at 0x680d52b0>)
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python3.8/site-packages/tornado/ioloop.py", line 605, in _run_callback
    ret = callback()
  File "/home/pi/oprint/local/lib/python3.8/site-packages/tornado/stack_context.py", line 277, in null_wrapper
    return fn(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/vendor/sockjs/tornado/transports/rawwebsocket.py", line 90, in <lambda>
    self.server.io_loop.add_callback(lambda: self.send_pack(message, binary))
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/vendor/sockjs/tornado/transports/rawwebsocket.py", line 85, in send_pack
    self.write_message(message, binary)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/tornado/websocket.py", line 251, in write_message
    message = tornado.escape.json_encode(message)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/server/util/tornado.py", line 42, in fixed_json_encode
    return json.dumps(value, default=JsonEncoding.encode, allow_nan=False).replace("</", "<\\/")
  File "/usr/local/lib/python3.8/json/__init__.py", line 234, in dumps
    return cls(
  File "/usr/local/lib/python3.8/json/encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/local/lib/python3.8/json/encoder.py", line 257, in iterencode
    return _iterencode(o, 0)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/util/json/__init__.py", line 33, in encode
    raise TypeError
TypeError

#### Did the same happen when running OctoPrint in safe mode?

No tested

#### Version of OctoPrint

Latest from staging/devel. OctoPrint 1.4.0rc3.dev9+g124f01ca running on OctoPi 0.15.0 (environment up to date).

#### Operating System running OctoPrint

Raspbian

#### Printer model & used firmware incl. version

Ender 3. Marlin 2.0.0 (released version)

#### Browser and version of browser, operating system running browser

N/A

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

Can provide if needed.

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-12-09:
Looks like it's trying to serialize something to JSON that it doesn't know how to serialize. Please provide the full log, it might contain hints as to WHAT it was trying to serialize.

gdombiak, 2019-12-09:
Thanks Gina. Will include full log file when I'm back home from work.

Thanks,
Gaston

foosel, 2019-12-09:
Best always include it right from the get go, chances are I'll need it ;)

gdombiak, 2019-12-10:
Hi Gina,

I included logs from yesterday and from today since it happened today too. Link to log: https://gist.github.com/gdombiak/99aab9590195d31f6f16ccdd249fa12f

In all cases, I see Palette 2 plugin right after the exception. I wonder if the plugin is the one trying to send some data type that is not serializable!?!?! Will see what else I can dig. Will share stack trace with Palette 2 devs in case they know about this.

foosel, 2019-12-10:
Pretty sure it's the Canvas plugin tbh based on the timings here. Took at look at the source that's available on gitlab, not sure if that's the current one, and can't see anything suspicious, but I have questions. How did you get that to run under Python 3.8 given that the plugin itself doesn't yet mark itself Py3 compatible? And: If you disable both palette related plugins, does the error go away?

foosel, 2019-12-10:
Oh, and: if you run things under Py2, does the same happen? Because I have a sneaking suspicion that it's caused by a bytes vs str issue under Py3...

gdombiak, 2019-12-11:
I updated and contributed back changes to run both plugins on Python 3. I think that you are onto something. One of the changes I had to do was around StringIO. Here are my 2 PRs I sent their way:
1. https://gitlab.com/mosaic-mfg/canvas-plugin/merge_requests/41/diffs#note_254538740
1. https://gitlab.com/mosaic-mfg/palette-2-plugin/merge_requests/78/diffs#note_254821706

I just disabled both plugins and running a few more prints. So far so good. Will report with latest update later tonight. My other OctoPrint instance runs Python 2 and OctoPrint 1.3.12 including those 2 plugins.   I do not have OctoPrint 1.4.0rc3.dev19+g5d1ad5d7 running on Python 2 to report that setup. Might not have time to create this setup.

gdombiak, 2019-12-11:
After many prints and many other tests .... I haven't see this error after I disabled the canvas & palette plugins. Contacted devs pointing them to this ticket. Still printing more things. Will reenable tomorrow and if error starts to appear again then will close this ticket.

foosel, 2019-12-11:
I'm pretty sure it's related to this line:

https://gitlab.com/mosaic-mfg/canvas-plugin/blob/master/octoprint_canvas/Canvas.py#L320

Something in the list of users can't serialize. So I'd suggest to investigate what kind of data is stored in there. The JSON encoder supports unicode and byte strings, int, float, dicts, lists, frozendicts and the internal User, Group and Permission classes. It's used for Flask, Tornado and SockJS JSON serialization and supports more data types than the stock implementations.

Could be that some kind of wrong data format is being dumped into the yaml that the data pushed is taken from, and that's causing this. I had one instance where binary blobs where written to yaml due to accidentally pushing bytes there, maybe something similar is happening here. In any case, it *strongly* smells like this is a problem specific to either one of these third party plugins and Py3, most likely the canvas one, and not a regression.

gdombiak, 2019-12-11:
Gina, great detective work!!! I pointed plugin devs to this thread so they will see your pointer and extra info. 

Later tonight will check logs again. If no error appeared then will close this ticket, ok?

osubuu, 2019-12-11:
Hi Gaston and Gina,

Thanks for looking into this. The `development` branch of both the Palette 2 and CANVAS plugins already contains a few Python2/3 compatibility fixes. Those branches were tested with  **Python 3.7.0 on OctoPrint 1.4.0rc2** and with **Python 2.7.16 on OctoPrint 1.3.12**. 

These changes weren't merged into the `master` branch yet, which explains why you ran into these issues.

@gdombiak If the `development` branches of both plugins work on your setup, agreed that this ticket can be closed.

Thanks!

gdombiak, 2019-12-11:
Hey @osubuu , I will download and install your development branch and give it a try tonight. Will report back findings here.

Thanks,
Gaston

gdombiak, 2019-12-12:
Updated to development branch of both plugins and no errors have appeared in the logs. Will keep testing but looks promising. Thanks @osubuu  and @foosel for your support. Closing ticket now.

closed 2019-12-12 by gdombiak