OctoPrint/OctoPrint #3376: Connecting serial 1000+ times via web interface or API crashes Tornado with MemoryError

MatthewCroughan, 2019-12-08:
#### What were you doing?

Connect the serial port 1000+ times via the api or by manually clicking on it in the browser page.

#### What did you expect to happen?

That Octoprint would not accumulate memory over time

#### What happened instead?

Octoprint accumulates virtual memory over time, leading to a MemoryError in Tornado.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint
1.3.12 (but happens in all versions)

#### Operating System running OctoPrint
Linux

#### Printer model & used firmware incl. version
Not relevant

#### Browser and version of browser, operating system running browser
Not relevant

#### Link to octoprint.log
Cannot provide at this time, details included in #3343 but bot nuked it. I need time to grab all the relevant logs.

#### Link to contents of terminal tab or serial.log
Cannot provide at this time, details included in #3343 but bot nuked it. I need time to grab all the relevant logs.

#### Link to contents of Javascript console in the browser
Cannot provide at this time, details included in #3343 but bot nuked it. I need time to grab all the relevant logs.

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

This is a very simple issue. Click the "connect" button upwards of 1000 times and Octoprint will crash.

What happens is that every time the serial port is attempted to be connected to, virtual memory is allocated which is never freed. Eventually Tornado will crash with MemoryError, this of course also applies to the API.

```
20.11.19 16:33:07 (+0000)  octoprint  2019-11-20 16:33:07,014 - tornado.general - ERROR - Uncaught exception
20.11.19 16:33:07 (+0000)  octoprint  Traceback (most recent call last):
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/octoprint/server/util/tornado.py", line 775, in _server_request_loop
20.11.19 16:33:07 (+0000)  octoprint      ret = yield conn.read_response(request_delegate)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/gen.py", line 1055, in run
20.11.19 16:33:07 (+0000)  octoprint      value = future.result()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/concurrent.py", line 238, in result
20.11.19 16:33:07 (+0000)  octoprint      raise_exc_info(self._exc_info)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/gen.py", line 307, in wrapper
20.11.19 16:33:07 (+0000)  octoprint      yielded = next(result)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/http1connection.py", line 159, in _read_message
20.11.19 16:33:07 (+0000)  octoprint      max_bytes=self.params.max_header_size)
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 256, in read_until_regex
20.11.19 16:33:07 (+0000)  octoprint      self._try_inline_read()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 711, in _try_inline_read
20.11.19 16:33:07 (+0000)  octoprint      pos = self._read_to_buffer_loop()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 625, in _read_to_buffer_loop
20.11.19 16:33:07 (+0000)  octoprint      if self._read_to_buffer() == 0:
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 738, in _read_to_buffer
20.11.19 16:33:07 (+0000)  octoprint      chunk = self.read_from_fd()
20.11.19 16:33:07 (+0000)  octoprint    File "/usr/local/lib/python2.7/site-packages/tornado/iostream.py", line 1052, in read_from_fd
20.11.19 16:33:07 (+0000)  octoprint      chunk = self.socket.recv(self.read_chunk_size)
20.11.19 16:33:07 (+0000)  octoprint  MemoryError
```

MatthewCroughan, 2019-12-08:
You can quickly test this yourself by using `glances`. Simply unplug your printer so that there is no way Octoprint can connect, then click connect over and over. You will quickly see Python begin to allocate virtual memory, this will never be freed and will lead to a crash. 

[image]

foosel, 2019-12-08:
As you correctly pointed out, this is a `MemoryError` raised inside Tornado, not OctoPrint. Tornado is a third party project. This should probably be reported upstream. I still might look into a workaround for this at a later date when I find some time, however given that someone running a DOS attack on their instance is somewhat a corner case and as mentioned it happens in Tornado from the looks of it it doesn't have a high priority at the current time.

In the meantime, as I told you on IRC after repeated warnings to improve on your tone and general behaviour towards others there and elsewhere, you are not welcome on this project anymore.

m-kozlowski, 2019-12-24:
Clicking button 1000 times is undoubtedly corner case, but I'd say that periodic api polling while the printer is off is quite possible.
I've hit that issue after installing OctoScreen ;)

StarWitch, 2020-04-06:
Getting this on a Raspberry Pi 3 running Octoprint 1.4 as well.

My use case was having the printer disconnected from OctoPrint, but still having OctoPrint online so I could use the webcam/Discord integration features. (Printing from the SD card at the time.)

Having it handle this gracefully would be a plus.

StarWitch, 2020-04-16:
Oh, I should also add: this happens within a matter of hours/minutes for me, not days. I have mostly stock settings as well. So it's possibly become a bigger issue than initially reported.

kantlivelong, 2020-05-06:
Confirmed this still occurs with Tornado 5.1.1 which is the latest release we will be able to use until Python 2 support is dropped.

DanielJoyce, 2020-05-07:
Actually, I suspect this spawning a bunch of features, one for each click. I don't think this is a tornado issue. You're asking tornado to undertake some action, it has no idea if that action will block in the backend, has latency, etc, so it just keeps firing futures.

What should happen is you keep the future for the first click.

If the first future has not resolved yet, then simply have the new request wait on it, Once the future resolves, clear the variable holding the future.

I'll need to dig in and see if this is the case.

foosel, 2020-06-18:
I found that when pressing connect with a connection handshake already in progress, OctoPrint would disconnect that and trigger a new one. That has been fixed on `maintenance` by ff47314d76aa0592ab888f8b8e898c708e16fcb8. I have not been able to generate a MemoryError, but I've not seen even a memory increase with this in place anymore. However I've been able to make Tornado crash due to a lack of free file handles when running a load test via [Artillery.io](https://artillery.io/). 

I've looked into the possibility of adding rate limiting inside OctoPrint/OctoPrint's tornado setup, but have found that to not be a good option. If (D)DOS attacks against the instances are to be feared, those should be mitigated a layer above, ideally on a separate physical system (considering the limited hardware that OctoPrint usually runs on).

Marking as solved since the originally reported memory-leak-on-connection issue is as much solved as I see possible from OctoPrint's side of things with the aforementioned changed, and anything else is beyond the scope of this project. If you require your printer host software to be able to serve thousands of requests per second, OctoPrint is not the right choice for you. And if there's third party software that spams OctoPrint with thousands of requests per second, that needs to be fixed on its end.

foosel, 2020-08-04:
1.4.1 has just been released

closed 2020-08-04 by foosel