OctoPrint/OctoPrint #3377: Pnotify CSS rule needs updating for fixed position

FormerLurker, 2019-12-08:
#### What were you doing?
I was testing my plugin and noticed that pnotify popups aren't always visible when popup dialogs are open.  These are absolutely critical for Octolapse because they show confirmation messages, and the results of several tests and other messages that must be visible for the UI to make sense.  My testing reveals that the css rule 
```
html>body>.ui-pnotify{position:fixed}
```
does not always select the pnotify dialog because it is not located within the div.page-container, and not a direct child of the body class.  I'm not sure exactly why this is, but it causes the above css rule not to work.

Steps to Reproduce:
1. Install OctoPrint 1.4.0rc3.dev9+g124f01ca or above (I tested in the latest staging/devel branch too, same result, but not 1.4.0rc2 for some odd reason)
2. Scroll down to the bottom of the page.
3. Open the javascript console and run the following command to open the settings dialog
```
$("#navbar_show_settings").click()
```
4. Navigate to the software update plugin and check for updates.  Notice that the popup is not visible.

#### What did you expect to happen?
The CSS rule setting the pnotify position to fixed should have applied to this and all pnotify popups.  I can't think of any reason not to just set 
```
div.ui-pnotify{position:fixed;}
```
in packed_libs.js instead of 
```
html>body>.ui-pnotify{position:fixed}
```
since all pnotify popups in all plugins need to play nicely with each other.  If any plugin tries to use position:fixed and others use position:absolute, the popups will overlap, which isn't good.

#### What happened instead?
pnotify dialogs are marked as position:absolute, which keeps them from showing up when scrolling down within the main dialog.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.4.0rc3.dev9+g124f01ca and the latest staging/devel branch.  Does not appear to be a problem in 1.4.0rc2.

#### Operating System running OctoPrint
Windows

#### Printer model & used firmware incl. version
NA

#### Browser and version of browser, operating system running browser
Chrome Version 78.0.3904.108 (Official Build) (64-bit)
Firefox 71.0 (64-bit)
Forefox Developer 72.0b3 (64-bit)

#### Link to octoprint.log
NA

#### Link to contents of terminal tab or serial.log
NA

#### Link to contents of Javascript console in the browser
NA

#### Screenshot(s)/video(s) showing the problem:
[image]
Notice that in the above image the PNotify popup is visible.  Now I'm going to scroll down on the main page (not in a popup)

[image]
Notice how the webpage is scrolled down about 50%.  Next I'm going to open a popup window.

[image]

See how the scrollbar on the right appears to be at the top?  In reality, we are scrolled down on the main Octoprint page, but the popup is scrolled to the top.  However, since pnotify popups are styled as absolute, they will stick to the top of the web page and not the top of the viewport.

I get the same behavior when using the OctoPrint settings menu (notice the rightmost scroll is disabled, as if the popup fills the entire page).  It's kind of hard to show that something is not showing up, but this is an image of the software update plugin that is NOT showing the 'check for updates' pnotify popup.  This is the results of my 'steps to reproduce' above.  Note that with octolapse installed, you can open the settings page when scrolled down a bit, which is not possible without triggering the settings popup in the javascript console as my steps to reproduce uses.

[image]

THANK YOU!  Let me know if you need any additional info, or further assistance debugging the issue.

I have read the FAQ.

foosel, 2019-12-09:
Ah, I know what's happening there. I had to move the PNotify popups into page-container since otherwise they'd stay visible even when logging out - a bug I found in RC2 and fixed on staging/devel. Looks like that had some side effects I'll have to take care of.

foosel, 2019-12-09:
Should be fixed by the above commit.

FormerLurker, 2019-12-09:
I've verified that this is fixed by the above commit.  Thank you so much!

CLOSED!

commit 6c92dd13 (2019-12-09): 🐛 PNotify needs position:fixed

Closes #3377
closed 2019-12-09 by FormerLurker