OctoPrint/OctoPrint #3378: System actions commands should have non-standard file descriptors closed before execution.

Artem-B, 2019-12-08:
#### What were you doing?

I've added a custom action to start/stop webcam daemon. The webacm control part worked fine, but octoprint started reporting errors:

```
2019-12-08 11:30:30,553 - octoprint.util.comm - ERROR - Unexpected error while connecting to serial port: /tmp/printer SerialException: '[Errno 11] Could not exclusively lock port /tmp/printer: [Errno 11] Resource temporarily unavailable' @ comm.py:_openSerial:2661 (hook default)
Traceback (most recent call last):
  File "/home/pi/OctoPrint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2661, in _openSerial
    serial_obj = factory(self, self._port, self._baudrate, settings().getFloat(["serial", "timeout", "connection"]))
  File "/home/pi/OctoPrint/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2650, in default
    serial_obj.open()
  File "/home/pi/OctoPrint/local/lib/python2.7/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/home/pi/OctoPrint/local/lib/python2.7/site-packages/serial/serialposix.py", line 312, in _reconfigure_port
    raise SerialException(msg.errno, "Could not exclusively lock port {}: {}".format(self._port, msg))
S
```

In the end I've tracked the problem down to the fact that for webcam daemon was holding printer's descriptor (`/dev/pts/0`) open and that apparently caused octoprint's attempt to obtain an exclusive lock to fail.

Add a command to config.yaml which would launch a long-running process in background. Something like this should do:
```
system:
  actions:
  - action: sleep
    command: sleep 100000&
    confirm: false
    name: just sleep
 ```

Restart octopi and then click on the `system->just sleep`.

#### What did you expect to happen?

Launched process should not inherit from octoprint any descriptors other than the standard ones.

#### What happened instead?

The launched commend inherited most of the open file descriptors from octoprint and that caused printer communication to fail.

When I checked the list of open descriptors for the launched process, it's obvious that it has a lot of open file descriptors inherited from octoprint:

E.g. here's what octoprint has open on my system:
```
octoprint 437   pi    0u      CHR    1,3      0t0   1030 /dev/null
octoprint 437   pi    1u      CHR    1,3      0t0   1030 /dev/null
octoprint 437   pi    2u      CHR    1,3      0t0   1030 /dev/null
octoprint 437   pi    3w      REG  179,2  4511020 135424 /home/pi/.octoprint/logs/octoprint.log
octoprint 437   pi    4u     IPv4  13112      0t0    TCP *:5000 (LISTEN)
octoprint 437   pi    5u  a_inode   0,11        0   6124 [eventpoll]
octoprint 437   pi    6r      CHR    1,9      0t0   1035 /dev/urandom
octoprint 437   pi    7r     FIFO   0,10      0t0  13094 pipe
octoprint 437   pi    8w     FIFO   0,10      0t0  13094 pipe
octoprint 437   pi    9u     IPv6  13113      0t0    TCP *:5000 (LISTEN)
octoprint 437   pi   10r  a_inode   0,11        0   6124 inotify
octoprint 437   pi   11w      REG  179,2        0 137277 /home/pi/.octoprint/logs/plugin_softwareupdate_console.log
octoprint 437   pi   12w      REG  179,2   585322 137278 /home/pi/.octoprint/logs/plugin_pluginmanager_console.log
octoprint 437   pi   13uW     CHR  136,0      0t0      3 /dev/pts/0
octoprint 437   pi   14w      REG  179,2    34484 128774 /home/pi/.octoprint/logs/serial.log
octoprint 437   pi   15u     IPv4  16542      0t0    TCP localhost:5000->localhost:43854 (ESTABLISHED)
octoprint 437   pi   16r     FIFO   0,10      0t0  66972 pipe
octoprint 437   pi   17w     FIFO   0,10      0t0  66972 pipe
octoprint 437   pi   18r     FIFO   0,10      0t0  66973 pipe
octoprint 437   pi   19w     FIFO   0,10      0t0  66973 pipe
```
and here's what web cam daemon ended up with:
```
webcamDae 2338   pi    0r      CHR    1,3      0t0   1030 /dev/null
webcamDae 2338   pi    1w      CHR    1,3      0t0   1030 /dev/null
webcamDae 2338   pi    2w      CHR    1,3      0t0   1030 /dev/null
webcamDae 2338   pi    3w      REG  179,2  4509350 135424 /home/pi/.octoprint/logs/octoprint.log
webcamDae 2338   pi   10r  a_inode   0,11        0   6124 inotify
webcamDae 2338   pi   11w      REG  179,2        0 137277 /home/pi/.octoprint/logs/plugin_softwareupdate_console.log
webcamDae 2338   pi   12w      REG  179,2   585322 137278 /home/pi/.octoprint/logs/plugin_pluginmanager_console.log
webcamDae 2338   pi   13u      CHR  136,0      0t0      3 /dev/pts/0
webcamDae 2338   pi   14w      REG  179,2    34484 128774 /home/pi/.octoprint/logs/serial.log
webcamDae 2338   pi   16r     FIFO   0,10      0t0  66972 pipe
webcamDae 2338   pi   17w     FIFO   0,10      0t0  66972 pipe
webcamDae 2338   pi   18r     FIFO   0,10      0t0  66973 pipe
webcamDae 2338   pi   19w     FIFO   0,10      0t0  66973 pipe
webcamDae 2338   pi  255r      REG  179,2     1672 137297 /home/pi/scripts/webcamDaemon
```

#### Did the same happen when running OctoPrint in safe mode?
N/A

#### Version of OctoPrint
1.3.12

#### Operating System running OctoPrint
```
% lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-12-09:
I actually put a fix for this into what is going to be 1.4.0 as part of #3212. Since I cannot control all possible calls to create subprocesses that might come from plugins, the serial port will now instead get the `close_exec` flag set on supported platforms which should prevent it from being passed on to subprocesses. 

However I'll also take a look at system commands behaving correctly in that regard in general for 1.4.1.

Artem-B, 2019-12-09:
@foosel Thank you. That should help.

Meanwhile, in case somebody else runs into this on older/current octoprint versions, one workaround is to close the descriptors manually in the launched script. E.g:

```
for fd in $(ls /proc/$$/fd/); do
    [ $fd -gt 2 ] && exec {fd}<&-
done
```

foosel, 2020-04-22:
The above patch changes things so we are now dropping file descriptors everywhere we can (if we can, Windows, python up until 3.7 & stdio redirection don't play well together)

commit e6336c62 (2020-04-20): ☂ Drop file descriptors on all subprocesses

At least if possible (close_fds + stdout/stderr redirection is not
supported on win32 in Python <= 3.7).

Closes #3378
closed 2020-08-04 by foosel