OctoPrint/OctoPrint #3380: Using Motion for multi-part jpeg stream fails to load snapshots and stream in the view

Valinwolf, 2019-12-10:
#### What were you doing?

1. Use motion instead of mjpg_streamer
2. Go to 'Control Tab'
3. See error

#### What did you expect to happen?

Show stream like mjpg_streamer

#### What happened instead?

Black screen with the text:
>Webcam stream not loaded
>
>It might not be configured correctly or require authentication. You can change the URL of the stream under "Settings" > "Webcam & Timelapse" > "Webcam" > "Stream URL". If you don't have a webcam you can also just disable webcam support there.
>Currently configured stream URL: https://REMOVED/stream

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

OctoPrint 1.3.12 installed from pip

#### Operating System running OctoPrint

ArchLinux ARM on Raspberry Pi 4 Model B, 4GB version

#### Printer model & used firmware incl. version

Not relevant

#### Browser and version of browser, operating system running browser

Brave 1.0.1 (Chromium 78.0.3904.108) (64-bit)
Windows 10 Pro

#### Link to octoprint.log

[Second file](https://gist.github.com/Valinwolf/9dcbccac59ccc8f7c437cf881051b722)

#### Link to contents of terminal tab or serial.log

Not relevant

#### Link to contents of Javascript console in the browser

[First file](https://gist.github.com/Valinwolf/9dcbccac59ccc8f7c437cf881051b722)

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

Yes

#### Works/Doesn't Work
✔️ mjpg_streamer under same setup (Nginx Proxy w/ SSL)
✔️ The stream URL test function
✔️ Plugins that use the stream URL (like The Spaghetti Detective)
✔️ Accessing the multi-part jpeg stream directly and proxied
✔️ Accessing the jpeg snapshot directly and proxied
❌ The snapshot URL test function
❌ The webcam view under the control tab

Valinwolf, 2019-12-10:
AH HA! I figured out what the problem is, however I do not know why it breaks motion, but not mjpg_streamer. The initial URL loaded has a query string appended with a bunch of numbers. This is the problem. I will figure out how to make Nginx ignore it and hopefully that will fix it.

Valinwolf, 2019-12-10:
Changed Nginx config, giving `proxy_pass` a trailing `/` and it works. I'm not sure why that fixed it or why mjpg_streamer doesn't require it, but it is fixed now.

closed 2019-12-10 by Valinwolf

foosel, 2019-12-10:
> The initial URL loaded has a query string appended with a bunch of numbers. This is the problem.

That's a simple cache buster to make sure we get a fresh copy of the stream each time (was needed to make stopping and restarting consumption of the stream actually work cross browser). I'm a bit surprised that motion hiccups on that tbh, but good to know you figured out and solved it!

TheFiZi, 2020-07-26:
I am having the exact same issue.

Can an option be added to the webcam configuration page called "disable cache buster"?

My `proxy_pass` has a trailing `/` and I still have the same issue.

The iOS OctoClient app has no problem loading the camera properly.