OctoPrint/OctoPrint #3381: [1.4.0rc2] /api/login?passive=true as sent by printoid causes Bad Request

foosel, 2019-12-10:
### Problem

RC2 fixed the 504 error for cura but still getting 'command error:api/login?passive=true(400)' every few seconds in printoid with the new 14RC1 or 14RC2

_Originally posted by @stevvie in https://github.com/foosel/OctoPrint/issues/3349#issuecomment-563496635_

foosel, 2019-12-10:
Issue is caused by printoid still using request parameters and a GET instead of a json body in a POST as specified by the API: http://docs.octoprint.org/en/master/api/general.html#post--api-login

1.3.x still supported that. JSON decoding behaviour in 1.4.0 changed due to an update of a library, making the fallback of reading request parameters no longer work.

Bringing back the fallback for 1.4.x but this should also be fixed on Printoid's side by adhering to the API specification!

foosel, 2019-12-10:
Solved by the above commit, confirmed to work with current Printoid LITE.

fieldOfView, 2019-12-10:
I was going to ask OP to fill in the issue template because there is info missing, but then I noticed who OP was ;-)

foosel, 2019-12-10:
I just needed a ticket number ;)

anthonyst91, 2019-12-10:
Hello Gina,

The issue is now fixed in Printoid (on the devel branch only) :
- OctoPrint v1.4.0-rc2 and above -> I now use POST with JSON body
- OctoPrint v1.3.12 and below -> still use POST with request parameters for retro compatibility 

I will publish the new release as soon as possible :)

foosel, 2019-12-10:
>     * OctoPrint v1.3.12 and below -> still use POST with request parameters for retro compatibility

That's actually not necessary, POST with JSON has always been the right thing, that the other stuff worked as well was just a left over from ... 1.1.x?

anthonyst91, 2019-12-10:
Right, just gave a try and indeed it is not necessary.

Previously I was using POST with query parameters (instead of JSON encoded
body) and it was running fine.

I will then totally replace this call by a POST with JSON body.

*Anthony STEPHAN*
Senior Android Developer at Renault-Nissan-Mitsubishi
Developer of Printoid for OctoPrint
Mobile:    +33 6 86 89 46 42
Email:
Website:
   contact@printoid.net
   https://printoid.net
<https://play.google.com/store/apps/details?id=fr.yochi76.printoid.phones.premium>
   <https://play.google.com/store/apps/dev?id=5122015473710814043>
<https://www.facebook.com/printoid/>   <https://twitter.com/androidprintoid>
   <https://www.linkedin.com/in/stephananthony/>
<https://www.patreon.com/printoid>

                     SUPPORT MY WORK ON PATREON !
<https://www.patreon.com/printoid>

foosel, 2019-12-12:
1.4.0rc3 has just been released

commit 0faa6879 (2019-12-10): ☂ Resilience against issue in Printoid

Closes #3381
closed 2019-12-12 by foosel