OctoPrint/OctoPrint #3382: [1.4.0 rc] Python 3 related issues

bzed, 2019-12-10:
#### What were you doing?

Run octoprint with python 3...

#### What did you expect to happen?

... it works...

#### What happened instead?

Various new backtraces ...

#### Did the same happen when running OctoPrint in safe mode?

yeah.

#### Version of OctoPrint

latest staging/devel git

#### Operating System running OctoPrint

Debian buster

#### Printer model & used firmware incl. version

not connected..

#### Browser and version of browser, operating system running browser

doesn't matter

#### Link to octoprint.log

I'll paste backtraces below.

#### Link to contents of terminal tab or serial.log

in bugs below

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

bzed, 2019-12-10:
I thought I should create an issue to collect all the Python3 related things I found so far. Otherwise I'll forget about them... Sorry for not creating more than one issue, fighting with that template is too annoying.

bzed, 2019-12-10:
**login using http basic auth is broken**

```
Traceback (most recent call last):
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/tornado/web.py", line 1489, in _execute
    result = self.prepare()
  File "/home/bzed/tools/OctoPrint/src/octoprint/server/util/tornado.py", line 257, in prepare
    self._fallback(self.request, b"")
  File "/home/bzed/tools/OctoPrint/src/octoprint/server/util/tornado.py", line 619, in __call__
    WsgiInputContainer.environ(request, body), start_response)
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/flask/app.py", line 1985, in wsgi_app
    response = self.handle_exception(e)
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/flask/app.py", line 1540, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/flask/app.py", line 1610, in full_dispatch_request
    rv = self.preprocess_request()
  File "/home/bzed/tools/OctoPrint/venv/lib/python3.7/site-packages/flask/app.py", line 1831, in preprocess_request
    rv = func()
  File "/home/bzed/tools/OctoPrint/src/octoprint/server/util/__init__.py", line 55, in loginFromAuthorizationHeaderRequestHandler
    if loginUserFromAuthorizationHeader():
  File "/home/bzed/tools/OctoPrint/src/octoprint/server/util/__init__.py", line 83, in loginUserFromAuthorizationHeader
    user = get_user_for_authorization_header(authorization_header)
  File "/home/bzed/tools/OctoPrint/src/octoprint/server/util/__init__.py", line 240, in get_user_for_authorization_header
    name, password = header.split(':', 1)
TypeError: a bytes-like object is required, not 'str'
```

I *think* the fix is

```
diff --git a/src/octoprint/server/util/__init__.py
b/src/octoprint/server/util/__init__.py
index 3e9a8c124..7b2dac664 100644
--- a/src/octoprint/server/util/__init__.py
+++ b/src/octoprint/server/util/__init__.py
@@ -237,7 +237,7 @@ def get_user_for_authorization_header(header):
        except TypeError:
                return None

-       name, password = header.split(':', 1)
+       name, password = str(header).split(':', 1)
        if not octoprint.server.userManager.enabled:
                return None
diff --git a/src/octoprint/server/util/__init__.py
b/src/octoprint/server/util/__init__.py
index 3e9a8c124..7b2dac664 100644
--- a/src/octoprint/server/util/__init__.py
+++ b/src/octoprint/server/util/__init__.py
@@ -237,7 +237,7 @@ def get_user_for_authorization_header(header):
        except TypeError:
                return None

-       name, password = header.split(':', 1)
+       name, password = str(header).split(':', 1)
        if not octoprint.server.userManager.enabled:
                return None
```

unfortunately something else fscked up the login, so I did not test it properly.

bzed, 2019-12-10:
**COMM autodetection is broken**
```
Traceback (most recent call last):
  File "/home/bzed/tools/OctoPrint/src/octoprint/util/comm.py", line 2694, in _openSerial
    serial_obj = factory(self, self._port, self._baudrate, settings().getFloat(["serial", "timeout", "connection"]))
  File "/home/bzed/tools/OctoPrint/src/octoprint/util/comm.py", line 2647, in default
    port = self._detect_port()
  File "/home/bzed/tools/OctoPrint/src/octoprint/util/comm.py", line 2627, in _detect_port
    programmer.connect(p)
  File "/home/bzed/tools/OctoPrint/src/octoprint/util/avr_isp/stk500v2.py", line 36, in connect
    self.sendMessage([1])
  File "/home/bzed/tools/OctoPrint/src/octoprint/util/avr_isp/stk500v2.py", line 102, in sendMessage
    checksum ^= ord(c)
TypeError: ord() expected string of length 1, but int found
```

struct.pack returns bytes in Python3, looping over bytes gives integers there.

I've tried
```
--- a/src/octoprint/util/avr_isp/stk500v2.py
+++ b/src/octoprint/util/avr_isp/stk500v2.py
@@ -98,7 +98,7 @@ class Stk500v2(ispBase.IspBase):
                for c in data:
                        message += struct.pack(">B", c)
                checksum = 0
-               for c in message:
+               for c in message.decode():
                        checksum ^= ord(c)
                message += struct.pack(">B", checksum)
                try:
```
But then the connections just timed out. There must be another issue hidden somewhere.

bzed, 2019-12-11:
Fix for the COMM autodetection - completely untested, just jumped into my mind (... in bed...):

```
diff --git a/src/octoprint/util/avr_isp/stk500v2.py b/src/octoprint/util/avr_isp/stk500v2.py
index 651ad5a4a..476d63088 100644
--- a/src/octoprint/util/avr_isp/stk500v2.py
+++ b/src/octoprint/util/avr_isp/stk500v2.py
@@ -99,7 +99,10 @@ class Stk500v2(ispBase.IspBase):
                        message += struct.pack(">B", c)
                checksum = 0
                for c in message:
-                       checksum ^= ord(c)
+                       if type(c) == int:
+                               checksum ^= c
+                       else:
+                               checksum ^= ord(c)
                message += struct.pack(">B", checksum)
                try:
                        self.serial.write(message)
```

foosel, 2019-12-11:
Both of these issues are fixed by the above commits. Keeping this open for now in case you come across anything else.

foosel, 2019-12-11:
Marked as a brainstorming ticket (don't have a "collection" label) and extracted the two issues into their own tickets. If you come across anything else, share here and I'll take a look and extract as necessary.

foosel, 2019-12-12:
Closing after all since you told me you'd open individual tickets for future issues ;)

closed 2019-12-12 by foosel