OctoPrint/OctoPrint #3383: [Request] Issue anti-reboot command before connecting to the printer

finasfinas, 2019-12-11:
- My printer runs on a Marlin clone.
- Octoprint talks with the printer via USB cable on /dev/ttyUSB0
- Everytime I press connect on Octoprint to connect to the printer the printer reboots
- If the printer is currently printing when pressing connect, the print job is halted and it is not possible to resume it.

It is possible to run a command on linux before connecting to the printer to avoid it rebooting as stated here:
https://community.octoprint.org/t/connecting-causes-printer-to-reboot/11341/4

the command for USB connected printers on ttyUSB0 is:
stty -F /dev/ttyUSB0 -hupcl 

So the much wanted feature request is:
Run the command every time before a connect to avoid rebooting the printer.

schumar, 2019-12-13:
As a quick fix, you could add that command to `/etc/rc.local`.

finasfinas, 2019-12-13:
adding to rc.local is not bullet proof, as if you disconnect the USB cable it stops working.
The only bulletproof solution is to send the command in every connect try.

schumar, 2019-12-14:
@finasfinas , you are of course right; if you want a bullet-proof system-side solution instead, you can use a udev-rule, which runs the `stty` whenever a new `ttyUSB` shows up; this will also correctly handle cases where it gets assigned another name, e.g. `ttyUSB1`.

finasfinas, 2019-12-14:
Indeed!!! that would clearly solve it. What do you think is the best way for octoprint to implement this, the udev rule you are suggesting or having the command to run on each connection try?

foosel, 2019-12-14:
Keep in mind that OctoPrint is cross platform and that's going to stay this way. There's no `stty` command or udev on all its target platforms. However a udev rule as part of Octo*Pi* might be interesting.

finasfinas, 2019-12-14:
@foosel should I close this feature request and open it elsewhere?

foosel, 2019-12-14:
Leave it open for now. Maybe there's a way to set the flags that stty sets in a cross platform way from python code. Any hints in that direction would btw be welcome.

schumar, 2019-12-15:
I also don't think that OctoPrint should set up udev-rules (which requires root-privileges, which it usually doesn't have); distributions (e.g. OctoPi) should take care of that.

It might be able to set the interface options in Python; after all, `stty` simply calls `ioctl`, which is available in Python too: [fcntl module](https://docs.python.org/3/library/fcntl.html)

```
$ strace -y -v -s 0 -eioctl stty -F /dev/ttyUSB0 -hupcl
[...]
ioctl(0</dev/ttyUSB0>, SNDCTL_TMR_STOP or TCSETSW,
    {c_iflags=0x500, c_oflags=0x5, c_cflags=0x8bd, c_lflags=0x8a3b, c_line=0,
     c_cc="\x03\x1c\x7f\x15\x04\x00\x01\x00\x11\x13\x1a\x00\x12\x0f\x17\x16\x00\x00\x00"
    }) = 0
```

Not sure which of those is important, but [tty_ioctl(4)](https://linux.die.net/man/4/tty_ioctl) and [termios(3)](https://linux.die.net/man/3/termios) might be useful.