OctoPrint/OctoPrint #3387: GCODE meta data from some slicers can't be calculated

AlbertMN, 2019-12-11:
#### What were you doing?
Trying to get the GCODE meta data _(dimensions, estimated print time, filament consumption & printing area)_ from files sliced through the "cura legacy" slicing built into Octoprint.

When the meta data is being generated for gcode files I sliced through Simplify3D, it works just fine - the data is generated correctly, but something seems to be wrong with the way it tries to get the meta data information from gcode that is sliced with cura legacy slicer.

Relevant files; https://www.dropbox.com/sh/laff90yt39j3gad/AABJdwlpRbbn_FRQLLccaM6Pa?dl=0
The Dropbox folder contains a `gcode` file that is sliced through Simplify3D _(wire-rail.gcode)_, a regular STL file, and then the gcode of that STL file, sliced through Octoprint with cura legacy _(soundbar-spacer.gco)_. The folder also contains the cura slicer settings used.

As said, with the Simplify3D-sliced file everything is fine, but not with the one sliced in cura. 

#### What did you expect to happen?
That the meta data gave me something like this; (filled out gcode analysis, no `null` values or `0` where there shouldn't be
```
gcodeAnalysis: {
  dimensions: {
    depth: 147.86,
    height: 4.8,
    width: 151.046
  },
  estimatedPrintTime: 273.8856498842895,
  filament: {
    tool0: {
      length: 65.1655000000001,
      volume: 0
    }
  },
  printingArea: {
    maxX: 156.046,
    maxY: 157.86,
    maxZ: 5,
    minX: 5,
    minY: 10,
    minZ: 0.2
  }
},
```

#### What happened instead?
The meta data I get is like this;
```
gcodeAnalysis: {
  dimensions: {
    depth: 0,
    height: 0,
    width: 0
  },
  estimatedPrintTime: 0.5344246021308753,
  filament: {
    tool0: {
      length: 0,
      volume: 0
    }
  },
  printingArea: {
    maxX: null,
    maxY: null,
    maxZ: null,
    minX: null,
    minY: null,
    minZ: null
  }
},
```
If the file is somehow corrupt / Octoprint is unable to get the data, I'd expect some sort of error. Even upon trying to analyse manually through a plugin using `octoprint.filemanager.analysis.QueueEntry`, setting the `priority` to `high`, the function returns `True`, and the log shows no errors, warnings or info as to why the data ended up looking like this.

I should also point out that sometimes the analysis doesn't generate the `printingArea` and `dimensions` info. The discovery of this bug started out with discovering this. More on that here; 

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.3.12 on OctoPi 0.15.1

#### Operating System running OctoPrint
Linux octopi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l

#### Link to octoprint.log
https://gist.github.com/AlbertMN/8d0902ff44dcee55016de9281990c114

I have read the FAQ.

AlbertMN, 2019-12-11:
A little update on this; I'm quite positive this only happens when slicing with the [Cura Engine Legacy slicer](https://github.com/OctoPrint/OctoPrint-CuraEngineLegacy) plugin for Octoprint.

[GCODE and stl file.zip](https://github.com/foosel/OctoPrint/files/3951295/tests.zip)

Attached is a zip containing;
- `soundbar-spacer-custom-sliced.gcode` - when importing this to Octoprint, the GCODE meta data is generated correctly _(sliced on Cura 15.04)_
- `soundbar-spacer-stl.stl` - when importing & slicing this _(using the same cura profile)_, the GCODE meta data calculation ends up "corrupt"

Also if you download the Octoprint-sliced `.gco` file, remove it from Octoprint and upload it manually, the data is still like this;
[image]

This issue might be better suited for the CuraEngineLegacy repository, although I still think that the data turning out as it does should be handled in some way by Octoprint - like an "MetadataAnalysisFailed" event being fired.

Let me know if I should open an issue in the Cura Engine repository instead.

foosel, 2020-04-24:
So looking into your first shared GCODE it's pretty empty:

```
;Generated from soundbar-spacer.stl (hash: 1d18c94028d91d02b069b117123e4a3aebe63329)
;Generated with Cura_SteamEngine 15.04.6
M107
G1 F2400 E-4.50000
G0 F9000 X0.000 Y0.000 Z5.000
M104 S0                     ;extruder heater off
M140 S0                     ;heated bed heater off (if you have it)
G91                                    ;relative positioning
G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way
M84                         ;steppers off
G90                         ;absolute positioning
;CURA_OCTO_PROFILE_STRING:eNrtWltv2zYYfTX6K/iWFks9SbabtIZetjV9WYECzbA2a0HQEm1xkUiBpOJckP++Q1K25cZGs67belEeEogiv+v5DvUx1GxuqWn0nGWclkLypchtkUbD8WDOJDU15zmt2GUaR9GgYlmBKTTj0nJNhaHXXKv0hJWGD8y50NaLoJlqpE1PdcMHmlvNMiuUpKzyw+PhZFBrIS21vKq5ZrbRfNwK0c6cGTOc2kJk55IbA1tGAyElFJqCl2WwCaMRxPDXS2Gz4vml1U2OGYtM5TydhlEy43bJuSS24CRrtIbZhLdTCZO5fyExsh49JMsCC7x9Qi4I5BSkakor6pKvZ5nhg+lpIQxx2gj+sjznOdTNleZe6OlD+ehB8KZieiFkOoG9pqlrhSAxuSh5+iQKE4QL5o74625quuFIjtaS5gIBQQh5Gk+6sa6EpHi44GUaI95LUUOEWiJCF6psKj99nV+fW9XYHRE2qhQ5LdkV3nVNeLLGQs5r2JtEW+qDgLGTMNOi6qIi2YThmubCWCYzDokwKCAIpouKlbTkcgHJ8cRJURdcl6ymecPKdnIrxI1QnxgDzelM2SJEzus0NcuQx3QEGc7feQP3Ci4WhYWYyYcJ+DDK3qAFq/36lUZ7VfPUSTcDY3ltKEBM4cJ2wnzMTJoM6pJZ4KKiLC+4t1EqyQcw1KqqDW2IF0KzSdSqIDZBzVQ1c76wsgyqmNCwjeLZ58r5xyWblRD1Ye2tgroJFFYkAw+fnEsj7FU6ggihjW1t8kCk8MUq7RESsGBVfUe6z0Fb3vEThNXUQrNSXPPWi7m4pIXSWsA42kjPBtDsIklZEJcp5TLDgDYxt+26OzyR7HsxWr1Qxn5JnIA8Qe6GErjMW4P+OHgnX8bRmLyOyK6f6doqRMUJUfO5WzGO9q7wEx0P5Z015KGYkyvVkAJ0QIR99E6+eBqTe/xMNQd2BVYhqsJlGu5jdUyeP47JySiKPrLaA8Q7D6CxyoWakZmwK6Z0mXYR9SFX19clPyRWEajl2AKIURWHB/41qN6gtLhXf/YDahdGTMibx0lE3rpfJzeB70It3ZJpBc4gZ6SpW50utyuT1ubwC2S2Ul5uckzeQFj0kaB4uW9+fOssRVkRZNSgKMwh7PWmOhA7AIBRV9Yv2RVSdzzeL9UxCbBsQpZfPN1vxZTNUIlg6+2sTG9qreAXOM8CqYvbg0OyQthp9HdB5hbFPTJ7ZH4byHSLkh7OPZy/Yzi7RaO+Bvoa+Fdq4H34gHd9Z9stCnSZx4NunNpe7s63e5rE6+4qRZD0FT64Ne82Oa4fS9Y9Z+iRXc/pexKmOev0rAUzNICTApxbDUOYhMZbSN/6bBZ1uplC1X7If6bTecku0AA9jA5Jc6B5rVl98GgA0+Gf65gDumgQeI92Z9bOUbM/eQaLhDwP3ZsqS7VctY9rQQE6aJSZV5fsezHa92K80ybDqaq5pMCquXP0sjmGiGHY0pveaYyP0fYy1xev25jpT8yIjBhuXYWZZ+RXBwES+uxn5CYgIjzekt8hEHNutgXfkhMEEMPdnvQWsl+5zBEHJ7xssYMH9+pkVVnwJw9Lg/OhF72tOkPLoH3RXZahWewuc89O7ssYJfG6VQYMdcGKYv9NohX3AlCVaPMc4YH9ltKTYWcucZH08qKna3n3lXVXzosk3hAD8IqQX7Cy4WarhveU7MvjZFP/vNPzOr2rFRUy6veKow1VINWh5fXUymRLGhsC20tSYdZZ1KGysx1zQLHxZLiPVD0ft0coJHeBiSdV5TxOyPMdxDV1R5J+VethTsJJktd0kmAfeT7avfmRUVU5Ep1DPYF12fknqCFswYQMyrYdclgGS6/PESqgni2w6Ujy68+/EJNpzqWLfnxEXrWThsNh+LzoS+yjJTb+DCUWStV9kd0VlXyarOh/r/yvoZxP93wkro7tbCi2BN9VK+b6ZxwQR7s5II4+Mwl4bY9vdh+XohROo3t57o9mvzbfewLsCbAnwPsRYPLdEmBP/T3199T/pVB/8hmpv99G/vNtZHQvShnD8G+NTO+3gY6+Qc/7DbTfQPdsoO/X15Yur7rXrY7a6zzhGlK4gtPdR92B+2Sg1DWnphC8XJ2jt9eWtm/n3Tl0TuPh0STcLmpvbrkLZ6sErK8KDXZugekk2kicl2rpbh9tH9GHa06di0Pu/w4u8uEqXLR92g29br+/cP+tcLBwl6i8N38Bo60gKA==
```

The problem is, I cannot reproduce it, when I slice the same file against the same version of CuraEngine and with your profile I get a full slicing result that also can get analysed.

An empty file obviously has no filament usage and no printing time or dimensions, so the results there are correct. The question is why your engine doesn't fill in the actual file, but that's probably rather an issue with the engine itself (which I don't maintain, only the plugin wrapped around it) and not OctoPrint. 

I loaded your second GCODE into OctoPrint and that calculated the metadata properly:

[image]

And I also just sliced the STL the same way and it worked:

[image]

So right now I have no explanation for what you are seeing, but what I can tell you is that at least based on the file you shared the analysis is correct: an empty file is an empty file, and getting a zero result for that is not a failed analysis. I suggest to circumvent OctoPrint completely as a test and try to slice the same file directly from the command line (you can see the command line used to invoke the slicer in the log, or if you enable logging to curaengine log). I'm pretty sure you'll get an empty file too, and if not that's definitely a ticket better suited for the CuraEngine Legacy Plugin's tracker.

I hope you don't mind that I close this for now as based on the current information there appears to be no bug in OctoPrint.

foosel, 2020-04-24:
Oh, I should note that I changed the analysis behaviour slightly in my build/on `maintenance` in that regard that it will still run the full analysis if it only gets a partial analysis from the slicer. The initial analysis of this ticket that found the GCODE file to be pretty much empty however happened before that, and print time and filament length were also still getting populated before.

closed 2020-04-24 by foosel