OctoPrint/OctoPrint #3395: Printer inactive on Octoprint 1.3.12 Marlin RC6

averter, 2019-12-17:
#### What were you doing?
Trying to 3D print and had just installed/configured Octoprint 1.3.12 on a Raspberry Pi 3. The 3D printer is a Prusa i3 rework with an old version of Marlin (RC6). 
1- Hit the Print button
2- The printer set the hotbed temperature
3- After a few seconds received a "Error reported by printer Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Heating failed, system stopped! Heater_ID: bed - Printer halted. kill() called!"

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?
Printer to respond to controls

#### What happened instead?
The printer is always inactive.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
Octoprint 1.3.12

#### Operating System running OctoPrint
OctoPi 0.17

#### Printer model & used firmware incl. version
Prusa i3 Rework, Marlin RC6

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log
See serial.log [here](https://gist.github.com/averter/ea9f55f8e67cff1b598b2aa1fb5026c6) instead 

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2019-12-17:
That's your firmware telling you that your bed heater is not behaving as expected and not an issue with OctoPrint. Make sure your printer is actually powered on.

closed 2019-12-17 by foosel

averter, 2019-12-17:
Well...I know how silly this is. However and in my defense, it only happened because this raspberry pi is new and had a fan attached to it, which is quite noisy, and made me believe that the printer was on (with the passive cooling fan). Thanks and sorry for the waste of time

foosel, 2019-12-17:
Don't worry, you aren't the first one to make this mistake or I wouldn't have immediately assumed this to be the problem 😅