OctoPrint/OctoPrint #3396: Time-lapse with libx264 lists incorrect (0 byte) file size in GUI

JohnOCFII, 2019-12-17:
[image]

#### What were you doing?

Using the new `libx264` to create a time-lapse.

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

Timelapse to be be created showing accurate file size.

#### What happened instead?

Timelapse shows 0 bytes file size.

#### Did the same happen when running OctoPrint in safe mode?

Not tested.

#### Version of OctoPrint

#### Operating System running OctoPrint

1.4.0rc3

#### Printer model & used firmware incl. version

Prusa MK3S with MMU2S

#### Browser and version of browser, operating system running browser

Safari 13.0.4 on macOS Catalina (10.15.2)

#### Link to octoprint.log

https://gist.github.com/JohnOCFII/a1763040858b31dbf7e33d75adcad107

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2020-01-16:
That turns out to actually have been a caching issue which is now fixed and ready for 1.4.0rc4. Thanks for reporting!

foosel, 2020-01-28:
1.4.0rc4 is out

commit e1a25a89 (2020-01-16): 🐛 Fix caching issue for freshly rendered timelapses

Closes #3396
closed 2020-01-28 by foosel