OctoPrint/OctoPrint #3400: [1.4.0rc3] JS-Function onDataUpdaterPluginMessage not ready after Events.CLIENT_OPENED

OllisGit, 2019-12-21:
#### What were you doing?
Tested my plugin "DisplayLayerProgress" for the first time with the RC 1.4.0.
Unfortunately the blue PrinterDisplay-PopUp in the Web-Browser doesn't appear any more.

My Plugin is listening for ```Events.CLIENT_OPENED``` on the server side. 
When the event occurs, it sends via ```self._plugin_manager.send_plugin_message``` a message to the web-browser. 
The javascript reacts on that message via ```self.onDataUpdaterPluginMessage``` and creates the blue PopUp PrinterDisplay.

As a workaround I implemented a thread.sleep before sending the data to the browser, because my assumption is that after the server-event ClientOpened appears, the javascript is not ready yet to receive message from the server.

```
# sleep needed for OctoPrint 1.4
import time
time.sleep(5)
self._plugin_manager.send_plugin_message(self._identifier, dict(initPrinterDisplay=True))
```
I created a small Sandbox-Plugin where you can reproduce the behaviour with 1.3.x and 1.4.x (https://github.com/OllisGit/OctoPrint-Sandbox)
It includes just the necessary stuff. 

If you install the plugin in 1.4.0 you can't see in the console a red message:
```Data from server received```
In 1.3.x it is visible.
After activating the sleep-function it also works in 1.4.0.

I am not sure about the consequences pausing the event-thread for 5 seconds. At the moment it looks like the "best" solution.

What do you think???

#### What did you expect to happen?
PrinterDisplay should be visible after user open or refresh the browser

#### What happened instead?
No PrinterDisplay appears

#### Did the same happen when running OctoPrint in safe mode?
not relevant.

#### Version of OctoPrint
1.4.0rc3

#### Operating System running OctoPrint
MacOS Development Virtual Printer

#### Printer model & used firmware incl. version
not relevant

#### Browser and version of browser, operating system running browser
Chrome

foosel, 2020-01-14:
I think it might be caused by race condition between authentication of the socket (before which no events will go through to the client - in 1.3.12 that was done by the ForceLogin plugin, in 1.4.0 it's built in and part of the new permission subsystem) and evaluation of the `CLIENT_OPENED` event.

The problem I now face is - I cannot reproduce this to validate this assumption, and that is extremely annoying. If you can reliably reproduce this locally, could you please try to only send the plugin message once you've also received a `USER_LOGGED_IN` event after a `CLIENT_OPENED` event? I'd hope this would fix it as well as the sleep. If it does, I'll look into further steps, which will probably involve another event `CLIENT_AUTHED` to be added, since changing the time when `CLIENT_OPENED` is triggered could cause breakage.

OllisGit, 2020-01-18:
Hi @foosel, 
I changed the behaviour in my sandbox-plugin as suggested (listening to  ```USER_LOGGED_IN```), but without success. It's reliable reproducible.

Event-Log from ```def on_event(self, event, payload):```:
```
octoprint.plugins.sandbox - INFO - Incoming: ClientOpened
octoprint.plugins.sandbox - INFO - Incoming: UserLoggedIn
octoprint.plugins.sandbox - INFO - Send MSG to client
--> the JavaScript-Side doesn't receive the MSG 
```
So, I still need to wait 5 seconds:
```
octoprint.plugins.sandbox - INFO - Incoming: ClientOpened
octoprint.plugins.sandbox - INFO - Incoming: UserLoggedIn
octoprint.plugins.sandbox - INFO - Sleeping for: 5s
octoprint.plugins.sandbox - INFO - Send MSG to client
--> the JavaScript-Side receive the MSG. You can see the console-output in the browser dev-tools
```
I didn't tested at on my Raspi3. I am testing on my MacBook pro. Maybe my Mac is tooooo fast ;-) So, the sleep-approach is currently the only solution.

From my perspective I will go with the workaround and if no other plugin-developer has such problem, you can close the issue.

BR
Olli

foosel, 2020-01-21:
I took another stab at this and then finally could reproduce it (I don't know what changed). In any case I've just pushed a change introducing a `ClientAuthed` event which will be fired once the socket has been authenticated and hence can actually be used to push data to a newly connected client. That should solve your issue without timeouts. I suggest to do something like:

``` python
def on_event(self, event, payload):
    self._logger.info("Incoming: " + event)

    if (not hasattr(Events, "CLIENT_AUTHED") and event == Events.CLIENT_OPENED) or (hasattr(Events, "CLIENT_AUTHED") and event == Events.CLIENT_AUTHED):
        # start sending messages
```

to remain backwards compatible to earlier versions.

Ready for 1.4.0rc4.

foosel, 2020-01-28:
1.4.0rc4 is out

closed 2020-01-28 by foosel

foosel, 2020-02-04:
So it turns out this was actually also a regression. I forgot that in the forcelogin plugin I had actually implemented a message buffer that would ensure messages that were sent before socket auth to still go through right after. That was missing in 1.4.0. I've fixed that now too, so even without plugin modifications things should now work just fine again.

commit 1d0829a1 (2020-01-21): 🐛 Introduce ClientAuthed event

May act as entry point for any kind of push messages a plugin might
wish to send to the client after a connect happened. Since depending
on the current permission configuration this might only be possible
once the client has authenticated on the socket, this is required.

Should close #3400