OctoPrint/OctoPrint #3401: Time estimate regressions?

rkarlsba, 2019-12-22:
Time estimate regressions

Time estimates are nice, but they are usually a bit lower than actual print times on my CR-10S printer. The same applies to the estimates of Cura, which usually are more off course than the ones of Octoprint. I just wonder if anything exists or could be made to do regressions for prints for particular printer types. If nn has a printer of make/model Z and print is estimated to x minutes, and ends up spending x*1.3 minutes, add that a log to allow for that to help further calculation of estimates. 

I'm not sure how this is practically possible, as in sorting printers by make/model by query (serial) or manual sorting and so on or if anything like this exists, but seems like good idea…

rkarlsba, 2019-12-22:
Feature request with "incomplete ticket" - heh - incomplete bot, prehaps ;)

fieldOfView, 2019-12-22:
The bot is doing what it intended to do; you removed the issue template so there's information missing in your ticket.

As to your issue, have you tried the Print Time Genius plugin? https://plugins.octoprint.org/plugins/PrintTimeGenius/

rkarlsba, 2019-12-22:
After exploring named plugin, I'll just close this. It'd be really nice to see such plugins getting merged into main, though

closed 2019-12-22 by rkarlsba