OctoPrint/OctoPrint #3403: Config.yaml appearance order not followed when replacing viewmodels

jneilliii, 2019-12-23:
#### What were you doing?

Reordering appearance > tabs in config.yaml using DragonOrder plugin. In this specific instance it was with another plugin that replaces the default Temperature tab (PlotlyTempGraph) that would not place itself where it was supposed to go based on the ordering defined in config.yaml.

#### What did you expect to happen?

Tabs to be re-ordered when interface loads.

#### What happened instead?

The replacement plugin's tab was always re-ordered to the end.

#### Did the same happen when running OctoPrint in safe mode?

Unable to reproduce in safe mode because issue requires a plugin that is replacing the core viewmodel.

#### Version of OctoPrint

1.4.0 RC3

#### Operating System running OctoPrint

Raspbian OctoPi 0.17

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Google Chrome Version 79.0.3945.88 (Official Build) (64-bit)

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

foosel, 2020-01-13:
Does this also happen in 1.3.12?

jneilliii, 2020-01-13:
I can confirm this happens in 1.3.12 as well.  Because the PlotlyTempGraph plugin incorporates accessViewModel I cloned the HelloWorld example and added the replaces for both settings and tab declaration like this.
```
	def get_template_configs(self):
		return [
			dict(type="navbar", custom_bindings=False),
			dict(type="settings", custom_bindings=False, name="Temperature", replaces="temperature"),
			dict(type="tab", name="Temperature", replaces="temperature")
```

Then used my DragonOrder plugin to re-order the tabs which resulted in the following config.yaml settings.

```
appearance:
  components:
    order:
      sidebar:
      - plugin_webcamSB
      - plugin_printer_safety_check
      - connection
      - state
      - files
      tab:
      - control
      - plugin_helloworld
      - gcodeviewer
      - terminal
      - timelapse
```

Upon reload of the interface the replaced "Temperature" tab was last in the UI instead of being after the control tab as expected.

[image]

Also, the replacement doesn't seem to work in Settings as expected at all.  The original temperature settings are still loaded and the Hello World setting is added in the bottom section under plugins. Not sure if replaces even works in settings or not.

foosel, 2020-01-14:
Ok. Then it's not a regression and will have to wait a bit until I have 1.4.0 taken care of. Marking accordingly :)

commit ff09cd0e (2020-04-07): 🐛 Fix ordering of replacement tabs

Order definition might contain either the plugin identifier or the
identifier of the component it replaces. Take that into account when
resolving order mapping.

Closes #3403
closed 2020-08-04 by foosel