OctoPrint/OctoPrint #3405: Memory leak when there are updates availalbe.

MFHFozzy, 2019-12-25:
I was chasing memory usage problems on my system. (Windows 10 Pro x64)
Opera was the biggest usages with 2.8GB of RAM. This is more than Assassins Creed Origins was using (2.6GB)!
By using the built in Opera task manager (shift-esc) i saw that is was one of the 2 Octoprint tabs i had running...and that tab had 4 pop up windows about plugin updates that i had not read.
Marking them "read" made the usage drop back down to 370K.

reloxx13, 2019-12-28:
i had not updates available.
had my win10 running al day with firefox and octoprint open

ended up by 5gb, 4gb was used by octoprint, nice.

MFHFozzy, 2019-12-29:
I have 2 tabs open, 2 different octopie tabs (2 different 3d printers), each has a camera. Only the tab that had pending updates was leaking. Although, that tabs printer was also turned off.

reloxx13, 2020-01-09:
Reopen this

foosel, 2020-01-13:
Yes sir.

foosel, 2020-01-13:
For the record, next time a "please" would be nice, I'm not a fan of being ordered to perform even valid requests in such a way, especially not when I get to read it the first day back in the office wading through a backlog from a nice and needed vacation ;)

MFHFozzy, 2020-01-14:
Welp, since its back....some more info:
2 tabs open, both have updates available, but tab 1 is not  eating memory up.
Tab 2 is, and thats the one where the 3d printer is turned off. 
Tab 2 has been open for < 4 hours, tab 1 has been open for days.
Tab 1 memory: 121,864 KB
Tab 2 memory: 425,464 KB

So the memory leak is real, but it may not be linked to the updates.
Going to apply updates and monitor again.

MFHFozzy, 2020-01-16:
24ish hours later:
Tab 1: 262,876 KB
Tab 2: 1,649,180 KB
Both up to date, no pending updates. Tab 2's machine is off seems to be the only difference?

reloxx13, 2020-02-13:
12hours
no updates, latest rc5

btw, the printer was off after ~6hours.
[image]

jrezin, 2020-04-01:
I believe my octoprint install is suffering from the same problem.

UI running overnight
[image]

running OctoPrint Version 1.4.0, on octopi 0.17.0

installed plugins:
* Cancel Objects (0.4.1)
* Dragon Order (0.1.4)
* Fullscreen Plugin (0.0.4)
* Gcodebar Plugin (0.1.5)
* GcodeEditor (0.2.8)
* LayerDisplay (0.4.1)
* Navbar Temperature Plugin (0.13)
* OctoKlipper (0.2.5-JEL-0.1)
* Python 3 Check (0.1.2)
* Stateful Sidebar (0.1.2)
* Tab Order (0.5.6)
* Themeify (1.2.0)
* TouchUI (0.3.14)
* Wemo Switch (0.1.5)

In a case like that, how to proceed to debug and find the culprit?

jrezin, 2020-04-04:
current, very basic, investigation points to
Tab Order (0.5.6) - /jneilliii/OctoPrint-TabOrder#33
as the culprit (at least) in my case... but I don't know how to properly debug a case like this, I'm just enabling and disabling plugins and waiting the memory consumption change.

jneilliii, 2020-04-04:
@MFHFozzy and @reloxx13, do you guys have Bed Level Visualizer or the Tab Order plugin's installed? Those are the two plugins of mine that have recently been identified as having memory/network leaks. The one with Bed Level Visualizer was related to the webcam loading and not disposing properly, I pushed an update that would only load the webcam while processing, but there is still an odd situation in Firefox that the network congestion remains even though the image is completely removed from the DOM. More details [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/208).

As for Tab Order it was because of a library I was using to be able to force tabs into the pull-down (hidden). I ended up going back and just patching the default bootstrap-taborder to have that functionality and it seems to have resolved it for me, additional testing by @jrezin expected. More details can be seen [here](https://github.com/jneilliii/OctoPrint-TabOrder/issues/33).

MFHFozzy, 2020-04-04:
I def had tab order installed, ill uninstall and check again

MFHFozzy, 2020-04-07:
Removed tab order and bed leveler:
after a couple days, 1 printer still off, the other has done a couple 
prints:
91.25 MB (05.62%) ++ top(http://192.168.0.92/#temp, id=1593)
43.55 MB (02.68%) ++ top(http://192.168.0.93/#temp, id=1599) (printer off)

I will reinstall tab order and test again, but so far problem solved, i 
think.

jneilliii, 2020-04-07:
Please join the thread over [here](https://github.com/jneilliii/OctoPrint-TabOrder/issues/33). Make sure to test with the latest version 0.5.7 using the URL below in plugin manager. I have yet to be able to reliably reproduce the memory leak with just Tab Order installed.

https://github.com/jneilliii/OctoPrint-TabOrder/archive/0.5.7.zip

foosel, 2020-04-08:
Am I understanding this correctly that this can likely be closed (at least for now)?

MFHFozzy, 2020-04-08:
Give me a couple more days to verify

jneilliii, 2020-04-08:
I've been unable to reproduce the memory leak with the latest patched version of TabOrder linked above and latest version of Bed Level Visualizer (with the exception of FireFox). But I also don't believe I've received an update notice on anything either since installing them, so the issue could still be tied to update notices in combination with these plugins maybe.

foosel, 2020-04-28:
Anything new here?

jneilliii, 2020-04-28:
I'm still at a loss. I haven't been able to reproduce the extremely large memory usage reported by others reliably. I do have a version of the plugin that removes what we think is the culprit 9knockout bound fontawesome iconpicker), which is linked below. 

https://github.com/jneilliii/OctoPrint-TabOrder/tree/0.5.7_iconpicker_removed

MFHFozzy, 2020-04-28:
Im sorry, i have not been able to get back to this as both printers have been printing PPE gear for covid efforts non-stop. 
I do plan on getting back to this as soon as i can.

Currently, I think we need to do more deep diving into the details of what/where/how memory is being used. 
Specifically, the major memory leak reporting seems to only show up in Chrome using its task manager. 
I could not replicate on Waterfox/Firefox. 
My next step will be to replicate the memory issue using Chrome (done), then re-test with the updated versions.

I probably will have time after Thursday.

foosel, 2020-06-05:
And again my question: Anything new here? :)

jneilliii, 2020-06-05:
So with the latest release of TabOrder people have reported that it resolved the high memory utilization for them, but I don't know if that is completely related to this issue or not.

foosel, 2020-06-12:
Hm... since I have never seen this myself and don't use TabOrder, I'm going to close it for now. If it shows up again we can always reopen.

closed 2020-06-12 by foosel