OctoPrint/OctoPrint #3409: 1.4.0rc3 Python 3 Timelapse "expected a bytes-like object, str found"

MSeal, 2019-12-31:
It appears there's an encoding issue in the release candidate for python 3. I didn't see issues in the related package about this but the `sarge` library appears to be expecting bytes and receiving strings (`b''.join(b)`). I didn't dig too deeply at the code to determine which library was at root fault, but given this is in a release candidate for python 3 I figured here was the probably source of the issue.

#### What were you doing?

1. Install the 1.4.0rc3 version in python with with all direct dependencies up to date in a dedicated venv for Octoprint using a Raspberry Pi 3B with the V2 Raspberry Camera
1. Use the following webcam settings (these links work and the test in octoprint UI settings page works for my setup):

```yaml
webcam:
    ffmpeg: /home/pi/mjpg-streamer/mjpg-streamer-experimental/mjpg_streamer
    snapshot: https://10.0.0.128/webcam/?action=snapshot
    snapshotSslValidation: false
    stream: https://10.0.0.128/webcam/?action=stream
    timelapse:
        fps: 25
        options:
            minDelay: 5.0
            retractionZHop: 0.0
        postRoll: 30
        type: zchange
    watermark: false
```

2. Start a print (any print will do)
3. See error (logs below)

```
2019-12-31 14:26:39,831 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Operational"
2019-12-31 14:26:40,448 - octoprint.timelapse - ERROR - Could not render movie due to unknown error
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/timelapse.py", line 901, in _render
    returncode, stdout_text, stderr_text = c.call(command_str, universal_newlines=True)
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint/util/commandline.py", line 209, in call
    all_stderr += process_stderr(p.stderr.readlines(timeout=0.5))
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/sarge/__init__.py", line 352, in readlines
    data = self.read(sizehint, block, timeout)
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/sarge/__init__.py", line 296, in read
    result = b''.join(b)
TypeError: sequence item 0: expected a bytes-like object, str found
```

#### What did you expect to happen?

Timelapse to function as expected.

#### What happened instead?

Timelapse feature failed with the error listed above.

#### Did the same happen when running OctoPrint in safe mode?

Yes. I don't have any extensions on in the fresh install.

#### Version of OctoPrint

1.4.0rc3

#### Operating System running OctoPrint

```bash
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

#### Printer model & used firmware incl. version

Probably not relevant, but Prusa Mk2 MMU firmware V3.2.3

#### Browser and version of browser, operating system running browser

Probably not relevant, but Chrome on Windows and on Ubuntu 18.04

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

[octoprint.log](https://github.com/foosel/OctoPrint/files/4013614/octoprint.log)

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

I have read the FAQ.

tedder, 2020-01-01:
What version of `sarge` do you have?

MSeal, 2020-01-01:
I noticed it hadn't updated in a long time. I have `sarge==0.1.5.post0` (latest) btw.

foosel, 2020-01-14:
So the issue here is the `universal_newlines` mode set on stdout/stderr for timelapse rendering specifically, because that enforces parsing as text. I guess Python 2 doesn't have this issue simply due to less strict type checking in anything str/byte related.

The question now is why the universal newline flag was even set on the line in question, and I'll try to dig that up. From the looks of it, things work just fine without it as well, so maybe it's really as easy as removing one single keyword argument (which is deprecated under Py3 anyhow).

MSeal, 2020-01-14:
Probably put in place for windows execution, but most video / picture loading application these days don't care about line endings so you might be right that it's unneeded.

foosel, 2020-01-15:
Sadly the rabbit hole goes deeper, see 4bf5984ace085a65d74570ffca2233cca7947e24 and https://stackoverflow.com/questions/7632589/getting-realtime-output-from-ffmpeg-to-be-used-in-progress-bar-pyqt4-stdout/35422274#35422274

So this is going to need some digging.

foosel, 2020-01-15:
Digging done, patch ready for 1.4.0rc4

foosel, 2020-01-28:
1.4.0rc4 is out

closed 2020-01-28 by foosel

MSeal, 2020-02-04:
I'll give this a try later in the week! Thanks for finding a fix :)

commit 482c2f71 (2020-01-15): 🐛 Fix timelapse progress under Py3

Closes #3409